OpenEtherCATsociety/SOEM #288: Propagation dealy measurement

mattharry, 2019-04-30:
Hi, I am using SOEM 1.3.1 and noticed that the propagation delay measurements vary in case of single and redundant mode, what is the factor which leads to this variation.
Each slave have two ports i.e 0 and 1. Setup remains the same throughout the observation, only difference is the redundant chain. Here is the test result.
Single Port

> [root@localhost red_test]# ./red_test p4p3  5000
SOEM (Simple Open EtherCAT Master)
Redundancy test
Starting Redundant test
ec_init on p4p3 succeeded.
3 slaves found and configured.
Slave:1 Name:? M:00000000 I:00000000 Output size:  0bits Input size:  0bits State: 4 delay:0.1
         Out:00000000,   0 In:00000000,   0
Slave:2 Name:? M:00000000 I:00000000 Output size:  0bits Input size:  0bits State: 4 delay:703.1
         Out:00000000,   0 In:00000000,   0
Slave:3 Name:? M:00000000 I:00000000 Output size:  0bits Input size:  0bits State: 4 delay:1441.1
         Out:00000000,   0 In:00000000,   0
Calculated work counter 0
Request operational state for all slaves
Operational state reached for all slaves.
Request safe operational state for all slaves        0, dt       -50000, O: I:
End redundant test, close socket
End program

Redundant Port

> [root@localhost red_test]# ./red_test p4p3 p4p4 5000
SOEM (Simple Open EtherCAT Master)
Redundancy test
Starting Redundant test
ec_init on p4p3 succeeded.
3 slaves found and configured.
Slave:1 Name:? M:00000000 I:00000000 Output size:  0bits Input size:  0bits State: 4 delay:0.1
         Out:00000000,   0 In:00000000,   0
Slave:2 Name:? M:00000000 I:00000000 Output size:  0bits Input size:  0bits State: 4 delay:572736008.1
         Out:00000000,   0 In:00000000,   0
Slave:3 Name:? M:00000000 I:00000000 Output size:  0bits Input size:  0bits State: 4 delay:332661750.1
         Out:00000000,   0 In:00000000,   0
Calculated work counter 0
Request operational state for all slaves
Operational state reached for all slaves.
Request safe operational state for all slaves        0, dt       -50000, O: I:
End redundant test, close socket
End program
 
As propagation delay is the time difference between leaving and returning frame which dosen't fits the redundant scenario, so as a workaround I tried to see if their is any difference if I close port1 of last slave so that the same frame will loop back (similar to non redundant mode) but the results were still different.

ArthurKetels, 2019-04-30:
The DC module in SOEM was never intended to be used in combination with redundant master ports. So some parts work, others (like propagation delay measurement) do not. Feel free to explain you use case where you would need both. Perhaps I will feel inclined to find a solution.

mattharry, 2019-05-02:
The use case is, if there is a need to maintain the redundancy so from the last slave instead of an open port, it loops back to master redundant port.
Well I found a fixture for this case and now propagation delay gets calculated in a similar way as it does in single port .

ArthurKetels, 2019-05-02:
Yes, I understand you want to have redundancy. But you have to realise that exact timing between slaves is no longer guaranteed. So either you have a single NIC and good timing or double NIC with slightly less timing accuracy. 

This timing problem is hardware limited in the master. EtherCAT itself and the slaves do not suffer from timing problems in redundant mode. So my question was, what is the use case where you need both very accurate timing and redundancy?

And it would be nice if you would share your solution to the problem so others can also learn.

mattharry, 2019-05-02:
` what is the use case where you need both very accurate timing and redundancy?`
For normal mode of operation, I require accuracy in timing. As data is crucial so don't want to drop packets that's why we are using redundant port as well.
I observed there is a slight increment in propagation delay with redundant configuration. 

It requires a minor change to fix the delay calculation
```
`/* Close port 1 of last slave*/
portSetting = 0xfc;
ecx_FPWR(context->port, 0x1000+(*context->slavecount), 0x101, sizeof(portSetting), &portSetting, EC_TIMEOUTRET);

/* DC calculation logic*/

portSetting = 0xf0;
ecx_FPWR(context->port, 0x1000+(*context->slavecount), 0x101, sizeof(portSetting), &portSetting, EC_TIMEOUTRET);
```
`

ArthurKetels, 2019-05-02:
Thanks for the patch information. 

This solution would work for the simplest case. I need to extend this a bit to be applicable for all cases. We can not assume the last slave has the connection to the redundant master NIC. And neither can we assume the port number on the slave. When we close the wrong port the algorithm would still fail.

mattharry, 2019-05-03:
Yes, as I said earlier this solution is specific for the setup on which I am working i.e each slave having 2 ports and the port1 of last slave connects to the redundant master NIC

mattharry, 2019-05-08:
Another thing is, I know it has been discussed earlier but still couldn't figure-out how master handles the time control loop filter adjustment for synchronization, and how its happening in stack, And how to synchronize all the ethercat slaves available on the chain.

ArthurKetels, 2019-05-08:
ec_configdc() does the work for you. It figures out what slaves have DC capability, and what are the propagation delays between each slave. It also sets all slaves local clock to the master clock, and appoints the reference slave ( the first DC slave). 

Each slave has its own local clock that is incremented by 10ns each tick. A hardware PLL in the slave compares the local clock with the reference clock (this is propagated by SOEM from the reference slave to all other slaves each PDO cycle). The output of the PLL generates 9ns, 10ns or 11ns each tick. This  way the local clock keeps synchronised with the reference slave.

As you noticed the master is only used to set the starting time. So it will drift in relation to the reference slave clock. Now there are two strategies, either propagate the master clock to the reference slave so the slave will sync on the master, or make your own PLL in the master to track the reference slave clock and adjust the master clock.

What I have found is that the master clock in most PC hardware has way more jitter and drift than the slave DC. So it makes sense to sync the master on the reference slave. An example is in red_test.c

nakarlsson, 2020-05-09:
Any progress on this issue?

nakarlsson, 2021-11-09:
close du to inactivity

closed 2021-11-09 by nakarlsson