OpenEtherCATsociety/SOEM #290: SOEM need Stack Size an Heap Size

lyhawk2007, 2019-05-05:
Soem platform USES STM32F429,I would like to know how much heap and stack configuration is appropriate.

ArthurKetels, 2019-05-05:
SOEM variables are mostly static assigned. When #ver1 is used no heap is needed. 2KB stack should do even when using printf.

The difference between #ver1 and #ver2 functions is the context structure. In legacy #ver1 it is static assigned and in #ver2 the user has to assign it at the appropriate sizes. This can be either static or dynamic on the heap. 

The maximum number of slaves determines the bulk of the structure. Then you can tweak the buffer sizes and the number of groups. When you disable concurrent CoE configuration (default now) you do not need dynamic tasks and a whole lot less stack.

From the beginning SOEM was designed to use little memory and be efficient on small controllers.

lyhawk2007, 2019-05-06:
After the SOEM stack is transplanted, a problem appears. The device will run for about 20 hours, and the system will crash. Is there any setting for the protocol stack?STM32F429 platform was used, and no real-time operating system was used.Stack = 0x10000, heap = 0x200

nakarlsson, 2019-05-06:
Do you run from a single thread? And when the crash occurs you only send/recv processdata? No parallel or ongoing CoE or other mailbox communication?

I see no reason why SOEM would "crash" the system after 20h in a single threaded sequential execution environment. When started it would be static as Arthur explaind.

lyhawk2007, 2019-05-06:
int ecat_main (void)
{
	uint32_t k=0,i,slave;
	ec_timet ec_timet_tt;
	void *ptr;
	test_osal_timer ();
	test_osal_current_time ();
	char ifname[]="test_eth0";
	printf("SOEM (Simple Open EtherCAT Master)\nSimple test\n");
	simpletest(ifname);
	while(1)
	{

		if((cyclic_callback_flag%5)==0)
		{
			ec_timet_tt=osal_current_time();
			cyclic_callback_flag=1;
			printf("OSTime=%d\r\n",OSTime);
			printf("current_time:sec=%d\r\n",ec_timet_tt.sec);
		}
		if(cyclic_callback_flag%2)
		{
			LED0_ON;
			set_output_bit(1,3,1); /* LED1 */
		}
		else
		{
			LED0_OFF;
			set_output_bit(1,3,0); /* LED1 */
		}
		read_io();
	}
   return (0);
}

void read_io (void)
{
   ec_send_processdata();
   ec_receive_processdata(EC_TIMEOUTRET);
}

lyhawk2007, 2019-05-06:
The above is my code. I only read and write in the main loop

lyhawk2007, 2019-05-06:
I'm not using the operating system, the whole program is just a while (1) main loop

nakarlsson, 2019-05-06:
The stack usage in above functions from SOEM should be static, but you have your own port, so I guess you should review your code first.
There are no specific settings exception what Arthur mentioned. When SOEM is started you should be able to readout the stack/heap usage from your system? If those values are within the limits and you run the above code, you should be OK.

nakarlsson, 2019-08-10:
Answer provided

closed 2019-08-10 by nakarlsson