OpenEtherCATsociety/SOEM #291: Coe PDOs Mapping and Sending Issues

farawayfromhome, 2019-05-07:
I tested PDOs with slaveInfo.c .The code in slaveInfo.c just added some mapping code, but an Error message appears.

Code here:
   /* initialise SOEM, bind socket to ifname */
   if (ec_init(ifname))
   {
      printf("ec_init on %s succeeded.\n",ifname);
      /* find and auto-config slaves */
      if( ec_config(FALSE, &IOmap) > 0 )
      {
    	 COE_Setup();
         ec_configdc();
         while(EcatError) printf("%s", ec_elist2string());
        ......
     }
   }

 void COE_Setup(void)
{
	int retval;
	uint8_t u8val;
	uint16 u16val;
	uint32 u32val;
	retval = 0;
	u8val = 0;

	retval += ec_SDOwrite(1, 0x1600, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

	u32val = 0x60600008;/* 0x6060:1/8bits operation mode  */
	retval += ec_SDOwrite(1, 0x1600, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	u32val = 0x60400010;/* 0x6040:0/16bits, control word */
	retval += ec_SDOwrite(1, 0x1600, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	u32val = 0x607A0020;/* 0x607A:1/32bits target Position  */
	retval += ec_SDOwrite(1, 0x1600, 0x04, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	u32val = 0x60B10020;/* 0x60B1:1/32bits Velocity offset  */
	retval += ec_SDOwrite(1, 0x1600, 0x05, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	u32val = 0x60B20010;/* 0x60B2:1/16bits Torque offset  */
	retval += ec_SDOwrite(1, 0x1600, 0x06, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
}
`

farawayfromhome, 2019-05-07:
information here:
[image]

[image]

[image]

[image]

ArthurKetels, 2019-05-08:
You modified the PDO mapping in the wrong slave state (SafeOP instead of PreOP).
It is easier to use the PO2SOconfig hook to reconfigure your slaves when needed.
See #160 and #94.

farawayfromhome, 2019-05-09:
I refer to #160 and #94 for mapping. But there is other problem: the slave can't switch to a safeop state with no error.
code here:
`/* initialise SOEM, bind socket to ifname */
   if (ec_init(ifname))
   {
      printf("ec_init on %s succeeded.\n",ifname);
      
      if(ec_config_init(FALSE) > 0)
      {
    	 if(ec_statecheck(0, EC_STATE_PRE_OP, EC_TIMEOUTSTATE*4) == EC_STATE_PRE_OP)
         {
    		printf("All Slaves in PRE_OP\n");
    	 }

    	 ec_slave[1].PO2SOconfig = COE_Setup;

    	 ec_config_map(&IOmap);
         ec_configdc();
        .....`
`int COE_Setup(uint16 slave)
{
	int retval;
	uint8_t u8val;
	uint16 u16val;
	uint32 u32val;
	retval = 0;
	u8val = 0;

	retval += ec_SDOwrite(1, 0x1C12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

	retval += ec_SDOwrite(1, 0x1600, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

	u32val = 0x609A0020;
	retval += ec_SDOwrite(1, 0x1600, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	u8val = 1;
	retval += ec_SDOwrite(1, 0x1600, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

	u16val = 0x1600;
	retval += ec_SDOwrite(1, 0x1C12, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);
	u8val = 1;
	retval += ec_SDOwrite(1, 0x1C12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
}
`
[image]

ArthurKetels, 2019-05-09:
Hmm, it looks like your slave crashed. The info above comes only from the eeprom in the ESC. According to the vendor id this is an Estun Automation servo drive. 

There is something weird with the CoE details. According to the bitmapping:
`#define ECT_COEDET_SDO       0x01`
`#define ECT_COEDET_SDOINFO   0x02`
`#define ECT_COEDET_PDOASSIGN 0x04`
`#define ECT_COEDET_PDOCONFIG 0x08`
`#define ECT_COEDET_UPLOAD    0x10`
`#define ECT_COEDET_SDOCA 0x20`
This slave with 0x1c has no SDO amd SDOINFO capabilities. How can you program a slave with PDOCONFIG without SDO? So this clearly must be a mistake.

To figure out what goes wrong please post a wireshark trace in pcap format here. So I can have a look wat goes on in detail. Perhaps we can find out the trouble spot. And if you can post the ESI file that would be great.

farawayfromhome, 2019-05-09:
I don't know how to 'post a wireshark trace in pcap format'. 
I run SOEM in RTEMS. I have just used wireshark to capture packets. The master , the slave and the PC are connected through the switch, but in this case, the master cannot find the slave.
I can post ESI file.

[ESTUN_ProNet_V205_RB100B0.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3161152/ESTUN_ProNet_V205_RB100B0.zip)

farawayfromhome, 2019-05-09:
So what should I do to capture packets through wireshark while testing SOEM?

ArthurKetels, 2019-05-09:
The easiest is to use your laptop and run both SOEM and wireshark there. SOEM runs on both windows and linux.

farawayfromhome, 2019-05-09:
Is there any other way? I am using RTEMS right now.  I never run SOEM on linux or windows before.

farawayfromhome, 2019-05-09:
Or is there any way I can config PDOs without SDO？

farawayfromhome, 2019-05-09:
The Engineer of Estun says that he can config PDOs with Twincat using the xml I post here.

farawayfromhome, 2019-05-10:
@ArthurKetels  Hi, are u still here?

I capture the packets successfully by wireshark .So I post here.I find out the AL error bit (0x130.4 = 1, 0x130.0~3 = 2, pre-op) has been set after sending SDOs to config PDOs.
Thanks.

[wireshark_cap(only ecat).zip](https://github.com/OpenEtherCATsociety/SOEM/files/3165700/wireshark_cap.only.ecat.zip)

closed 2019-05-15 by farawayfromhome

9876ncsystems, 2022-12-06:
Hello,
For us is not clear was the problem with ESTUN drives solved or not.
Now we have an ESTUN drive and controlling with SOEM is quite actual for us.
It looks like there is a similar problem as described above.
Regards!

ArthurKetels, 2022-12-06:
Did you actually try to operate the slave with SOEM? The above was as much a problem with the SOEM application as the slave firmware. Who knows problems are solved by new Estun firmware? Slaves that pass the mandatory slave conformance test with the CTT from the ETG have normally no problem with SOEM. It is when vendors only test against TwinCAT some problems may remain hidden.

9876ncsystems, 2022-12-06:
Yes.
Last 3 days we are trying to do this, as you can see trace of slaveinfo looks very “hopeful“.
[Slaveinfo_ESTUN_ED3L.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10165330/Slaveinfo_ESTUN_ED3L.txt)

Nevertheless at last we were able to control them with our test application (base on SOEM simpletest) in DC mode. In free run mode  devise is going in OP and we see death loop: No Fault -> (SW=0x6)-> Ready ->(SW=0x7)-> Wait to Switch ON->(SW=0xF)-> Stop at Fault /instead of Running/ -> No Fault. If in state Wait to Switch ON we are switching cyclic data transfer the drive close and keep the position loop.
We are palling to investigate what happened with TwinCAT. 
Regards!