OpenEtherCATsociety/SOEM #293: Ubuntu 18.04 no slaves found

cplasberg, 2019-05-09:
I'm using a Beckhoff CU1128 with connected ANYdrives. All works fine when using Ubuntu 16.04 but with Ubuntu 18.04 I can not find any slaves.
To validate I used `$ sudo ./slaveinfo [iface] -map`
I already had a look at [#172](https://github.com/OpenEtherCATsociety/SOEM/issues/172) and [#175](https://github.com/OpenEtherCATsociety/SOEM/issues/175) but did not find them helpfull for my situation.
Here's the Wireshark log from the 18.04 machine:
1	0.000000000	Private_01:01:01	Broadcast	ECAT	29	'BWR': Len: 1, Adp 0x0, Ado 0x103, Wc 0 
2	0.007938393	Private_01:01:01	Broadcast	ECAT	29	'BWR': Len: 1, Adp 0x0, Ado 0x120, Wc 0 
3	0.015926002	Private_01:01:01	Broadcast	ECAT	29	'BWR': Len: 1, Adp 0x0, Ado 0x120, Wc 0 
4	0.023935042	Private_01:01:01	Broadcast	ECAT	30	'BRD': Len: 2, Adp 0x0, Ado 0x0, Wc 0 
5	0.031939751	Private_01:01:01	Broadcast	ECAT	30	'BRD': Len: 2, Adp 0x0, Ado 0x0, Wc 0 
6	0.039942066	Private_01:01:01	Broadcast	ECAT	30	'BRD': Len: 2, Adp 0x0, Ado 0x0, Wc 0 
7	0.077657915	03:01:01:01:01:01	Broadcast	ECAT	60	'BWR': Len: 1, Adp 0x5, Ado 0x103, Wc 5 
8	0.077664346	03:01:01:01:01:01	Broadcast	ECAT	60	'BWR': Len: 1, Adp 0x5, Ado 0x120, Wc 5 
9	0.077665450	03:01:01:01:01:01	Broadcast	ECAT	60	'BWR': Len: 1, Adp 0x5, Ado 0x120, Wc 3 
10	0.077666607	03:01:01:01:01:01	Broadcast	ECAT	60	'BRD': Len: 2, Adp 0x5, Ado 0x0, Wc 5 
11	0.077667835	03:01:01:01:01:01	Broadcast	ECAT	60	'BRD': Len: 2, Adp 0x5, Ado 0x0, Wc 5 
12	0.077669088	03:01:01:01:01:01	Broadcast	ECAT	60	'BRD': Len: 2, Adp 0x5, Ado 0x0, Wc 5 

Thanks for your help!

ArthurKetels, 2019-05-09:
Perhaps related to https://github.com/OpenEtherCATsociety/SOEM/issues/203#issuecomment-417306167?
What is your kernel version?
Otherwise post a full wireshark capture (just raw .pcap file).

ArthurKetels, 2019-05-09:
Looking more close at the timing you posted perhaps it has to do with packet coalescing?
Look at https://github.com/OpenEtherCATsociety/SOEM/blob/master/drvcomment.txt

cplasberg, 2019-05-09:
Thanks for your quick response!
Kernel Version is 4.15.0-48. I also tried to fix possible packet coalescing with your suggested fix but without result.
Here is the Wireshark capture.
[slaveinfo_18.04.pcap.gz](https://github.com/OpenEtherCATsociety/SOEM/files/3162544/slaveinfo_18.04.pcap.gz)

ArthurKetels, 2019-05-09:
The issue is that packets only return to the master application after 70ms. SOEM tries to send some frames times out and finally just gives up. After that the responses (that are correct by the way) come in. 

So either the packets are buffered for transmission and not send over the wire until 70ms, or the network driver buffers incoming packets and releases them to the application after 70ms.

Try to find out what the changes are in the NIC driver. There must be some way of beating that thing into behaving. Did you check all possible options to tune your driver? Or do you have a way to print the current NIC settings?

cplasberg, 2019-05-10:
I had a closer look at the NIC settings and compared both machines.

Comparing the output of `sysctl -a|grep net.core` showed only one difference: `default_qdisc`.
On the 16.04 machine it was set to `pfifo_fast` while on 18.04 to `fq_codel`.

Changing it to `pfifo_fast`  with 
`sysctl -w net.core.default_qdisc=pfifo_fast`
solved the problem for me.

To make changes permanently you have to edit `/etc/sysctl.conf` and run `sysctl -p` afterwards.

@ArthurKetels Thanks for your help and the right hints!

closed 2019-05-10 by cplasberg

crown133, 2024-04-19:
hello, I'm also using the Beckoff cu1128 switch with SOEM, but it doesn't work. How to configure the cu1128 ? Could you give me a light guide? Thank you a lot!

ArthurKetels, 2024-04-19:
Please open your own issue instead of using an old one. 

More information is also welcome. How do you expect people to help your when you provide so little background information?

The cu1128 does not need any configuration. Connect one master to whatever port, connect slaves to other ports, and it works. There are no known issues with SOEM using this switch.