OpenEtherCATsociety/SOEM #295: LAN9252 SyncManager Watchdog error when moving to OP state

mikhail-urbant, 2019-05-10:
Dear developers,

I'm currently trying to solve a problem with a slave unable to move to OP state from SAFEOP.  This is a slave based on LAN9252 with fixed PDO mapping, generated with Beckhoff SSC tool. After trying to move to OP state I get "Sync Manager Watchdog error". From what I observe it seems like PDO data communication is failing for some reason. 

Do I understand correctly that no special PDO configuration is required for slaves with fixed PDO mapping? I have 16 bit output and 16 bit input which I want to access.

The output of slave_info:

> Slave:1
 > Name:ECAT1616
>  Output size: 16bits
>  Input size: 16bits
>  State: 4
>  Delay: 0[ns]
>  Has DC: 1
>  DCParentport:0
>  Activeports:1.0.0.0
>  Configured address: 1001
>  Man: 00000aca ID: 00009252 Rev: 00000001
>  SM0 A:1000 L: 128 F:00010026 Type:1
>  SM1 A:1080 L: 128 F:00010022 Type:2
>  SM2 A:1100 L:   2 F:00010064 Type:4
>  SM3 A:1400 L:   2 F:00010020 Type:3
>  FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:02 Act:01
>  FMMU1 Ls:00000002 Ll:   2 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
>  FMMUfunc 0:1 1:2 2:3 3:0
>  MBX length wr: 128 rd: 128 MBX protocols : 0e
>  CoE details: 23 FoE details: 01 EoE details: 01 SoE details: 00
>  Ebus current: 0[mA]
>  only LRD/LWR:0
> End slaveinfo, close socket
> End 

The output of simple_test:

> SOEM (Simple Open EtherCAT Master)
> Simple test
> Starting simple test
> ec_init on \Device\NPF_{B080920C-968C-4573-AC89-7ED2AA42CC93} succeeded.
> ec_config_init 0
> 1 slaves found and configured.
> Found ECAT1616 at position 0
> Current state = 1
> ec_config_map_group IOmap:001A25A0 group:0
>  Slave 1, configadr 1001, state 02
>   CoE Osize:16 Isize:16
>      ISIZE:16 16 OSIZE:16
>   SM programming
>     SM2 Type:4 StartAddr:1100 Flags:00010064
>     SM3 Type:3 StartAddr:1400 Flags:00010020
>   OUTPUT MAPPING
>     FMMU 0
>       SM3
>     slave 1 Outputs 001A25A0 startbit 0
>  =Slave 1, INPUT MAPPING
>     FMMU 1
>       SM2
>     Inputs 001A25A2 startbit 0
> IOmapSize 4
> Current state = 2
> Slaves mapped, state to SAFE_OP.
> Current state = 4
> segments : 1 : 4 0 0 0
> Request operational state for all slaves
> Calculated workcounter 3
> Not all slaves reached operational state.
> Slave 1 State=0x14 StatusCode=0x001b : Sync manager watchdog
> 
> Request init state for all slaves
> End simple test, close socket
> End program

I'm attaching Wireshark log of PDO access frame. It seems like Working Counter is not incremented in response frame.
[pdo_log.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3165206/pdo_log.zip)

The xml file describing the slave, generated by SSC tool:
[PIC32_LAN9252_.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3165221/PIC32_LAN9252_.zip)

We also tried to use TwinCAT which successfully moves the slave to OP.

Thank you for your help!

nakarlsson, 2019-05-10:
Check for SSC 5.12 patches concering SM types and 0x1C00 objects.

mikhail-urbant, 2019-05-10:
> Check for SSC 5.12 patches concering SM types and 0x1C00 objects

Do you mean that SSC 5.12 could have introduced some bugs as it was discussed in other topics already?

We are using SSC 5.12.

Thank you!

nakarlsson, 2019-05-10:
https://github.com/OpenEtherCATsociety/SOEM/issues/214

mikhail-urbant, 2019-05-10:
With your help I was able to solve this problem. 

Indeed the PDO process was failing and the reason was that slave was incorrectly reporting SM2 SM3 directions. For now I modified SOEM so it would look at EEPROM at the first place for SyncManager mapping. After that I was able to move the slave to OP state, and PDO access started to function normally.

Thank for your help, nakarlosson.

closed 2019-05-10 by mikhail-urbant

WangKaimeng, 2020-05-21:
> SyncManager mapping

I met the same issue these days. Could you please tell me how to modify the SOEM code? Thanks in advanced!

gvcode1, 2021-08-22:
I meet the same problem here, from SAFEOP TO OP failed, and report this watchdog error alstatus 0x001B, could you share some piece code of your modification about this issue? @mikhail-urbant