OpenEtherCATsociety/SOEM #296: Timing issues between a SOME Ethercat Master and Ethercat motion controller

robotixdeveloper, 2019-05-14:
We are implementing a motion control system based on Ethercat with SOEM and industrial motion controllers as Ethercat Slaves.
Our Ethercat motion slaves are using the “cyclic position mode”. This means they require a position update on every clock tick of the Ethercat bus. If a step is missed the slaves are reporting an error and stop.
 
We have based our Ethercat Master on the Red_Test.c example (you find the code below). We are running it on Linux Ubuntu 16.04. LTS without any real time patch. Our ethercat cycle time is 1 ms.
As HW we use an Intel NUC (NUC7I5BNH) with a 4 core i5 CPU and Intel® Ethernet Connection I219-V.
 
Currently we are facing the problem that our motion controller slaves are not receiving all packages on time. They required on every clock tick a PDO and are reporting a “missing PDO” after a couple of seconds. 

Could you please let us know:
 
* If the example we implemented is current or any other synchronization mechanism need to be implemented
* A real time kernel is required
* If a real time kernel is required, which kernel to choose for Ubuntu 16.04 LTS and 18.04 LTS

struct timespec   ts, tleft;
int ht;
int64 cycletime;

clock_gettime(CLOCK_MONOTONIC, &ts);
ht = (ts.tv_nsec / 1000000) + 1; /* round to nearest ms */
ts.tv_nsec = ht * 1000000;
cycletime = 1000000; /* cycletime in ns */

toff = 0;

/* cyclic loop */
int32_t pos = 0;
int direction = 1;

for(i = 1; i <= 20000; i++)
{
   {
	 /* calculate next cycle start */
	 add_timespec(&ts, cycletime + toff);
	 /* wait to cycle start */
	 clock_nanosleep(CLOCK_MONOTONIC, TIMER_ABSTIME, &ts, &tleft);

	 wkc = ec_receive_processdata(EC_TIMEOUTRET);

	 if (ec_slave[0].hasdc)
	 {
		/* calulate toff to get linux time and DC synced */
		ec_sync(ec_DCtime, cycletime, &toff);
	 }
	 
	   if (i>10000) direction = -1;
	   
	   pos += direction;
	   setPosition(ec_slavecount,pos/5);
	ec_send_processdata();

   }
}

ArthurKetels, 2019-05-15:
No need to go so complicated. In your software example you get already your ec_DCtime variable that is updated each ec_send_processdata() cycle. This variable is the 64 timestamp the moment the PDO packet arrived at the reference DC slave. 

So you have the exact timestamp in DC time of your previous PDO exchange. You compare this to ts which is your PC reference time for the cycle. Just make a histogram of all deviations. ec_sync does the same to control toff, which is the offset from PC time to DC slave reference time. Plotting toff will give you the clock drift of PC time relative to DC reference.

robotixdeveloper, 2019-05-17:
Thanks for your reply, I have created a histogram. 
- 90% of all values are around 1ms which is perfect
- 5% are much faster 
- 5% are above 3m with sometimes one package with around 15 ms
It seems that something on the system is causing a severe delay of 15ms

How can this be avoided? The system has currently no application load.

robotixdeveloper, 2019-05-17:
[wiresharkcapture.pcapng.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3191344/wiresharkcapture.pcapng.zip)
You find attached the wireshark data

ArthurKetels, 2019-05-17:
Histo looks like normal for a standard non realtime kernel. So I can recommend to build your own preempt-rt kernel. Easiest to just take the latest from www.kernel.org. And use the same build settings as your ubuntu version. 

See : https://wiki.linuxfoundation.org/realtime/documentation/howto/applications/preemptrt_setup

robotixdeveloper, 2019-05-17:
I just started the build and will report the success. Do I need to configure the kernel afterwards in a special way or can I simply re-run my application?

robotixdeveloper, 2019-05-17:
The problem is solved. RT Patch and various settings on the system did the trick. Now things are running nicely. Thanks for your support!

closed 2019-05-17 by robotixdeveloper

miscz, 2019-08-07:
Hello!

I am facing a similar problem like robotixdeveloper. Running in cyclic sync position mode I am loosing packages and the servo is making cracking noises.
My setting is alo very similar: It is a Lenovo T460s with an core i5 with an Intel Ethernet Connection I219-V. Following the instruction from this post and others I did the following:

- patched my ubuntu 16.04 with the preempt rt 5.2.0 kernel
- added a function `set_realtime_priority()` (see below) in order to set the rt thread ecatthread from the red_test example to priority 99

```
void set_realtime_priority() {

     int ret;

     // We'll operate on the currently running thread.
     pthread_t this_thread = pthread_self();

     // struct sched_param is used to store the scheduling priority
     struct sched_param params;

     // We'll set the priority to the maximum.
     params.sched_priority = sched_get_priority_max(SCHED_FIFO);

     std::cout << "Trying to set thread realtime prio = " << params.sched_priority << std::endl;

     // Attempt to set thread real-time priority to the SCHED_FIFO policy
     ret = pthread_setschedparam(this_thread, SCHED_FIFO, &params);
     if (ret != 0) {
         // Print the error
         std::cout << "Unsuccessful in setting thread realtime prio" << std::endl;
         return;
     }

     // Now verify the change in thread priority
     int policy = 0;
     ret = pthread_getschedparam(this_thread, &policy, &params);
     if (ret != 0) {
         std::cout << "Couldn't retrieve real-time scheduling paramers" << std::endl;
         return;
     }

     // Check the correct policy was applied
     if(policy != SCHED_FIFO) {
         std::cout << "Scheduling is NOT SCHED_FIFO!" << std::endl;
     } else {
         std::cout << "SCHED_FIFO OK" << std::endl;
     }

     // Print thread scheduling priority
     std::cout << "Thread priority is " << params.sched_priority << std::endl;
}
```

- changed the TG3 settings of my ehternet adapter like mentioned in the drvcomment.txt (`ethtool -C eth0 rx-usecs 0 rx-frames 1 tx-usecs 0 tx-frames 1`)
- added time measuring to the rt thread to measure the cycle time (this does not have an effect on the timing behavior of the thread):

```
clock_gettime(CLOCK_MONOTONIC, &tspec_now);
cycletime[i] = (tspec_now.tv_sec*NSEC_PER_SEC + tspec_now.tv_nsec)-(tspec_b4.tv_sec*NSEC_PER_SEC + tspec_b4.tv_nsec);
ec_send_processdata();
clock_gettime(CLOCK_MONOTONIC, &tspec_b4);
```

The test program runs for 20 seconds with a cycle time of 2 miliseconds. I made some plots of the cycle time with its histogram
[image]
[image]

of `toff` with its histogram
[image]
[image]

and of `ec_DCtime`.
[image]

You can also find the wireshark capture of this test here:
[test3.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3475731/test3.pcapng.txt)

Any suggestions what might be wrong with my code or the setup? Is it correct, that `ec_DCtime` has a modulo behavior? Limited range?

Thank you for your help!

miscz, 2019-08-14:
Hello @robotixdeveloper !

I am facing the same problem. Can you tell me which kernel you used and which settings you made in the kernel config before you built it? Maybe you can upload the your `.config` file?