OpenEtherCATsociety/SOEM #297: CoE PDOs' subIndex Issues

farawayfromhome, 2019-05-15:
Hi all,
I have some problems with PDOs configuration.
There is a PDO whose sub-index 0 does not require PDO mapping, and sub-indexes 1 and 2 need to be mapped. Then when I use sdowrite to configure this PDO, how should I call it to print the subindex 1 and subindex 2 information in the slavemap. 
Just like this:
[image]

farawayfromhome, 2019-05-15:
I am very confused how they print out the information of subIndex1 and subIndex2.

ArthurKetels, 2019-05-15:
I am also confused what your question is. The above print out looks normal.

In general. An SDO consist of an index and a subindex. 

If the datatype of the index is not ARRAY or RECORD (or string) then only subindex 0 exists and that is the one you want to map as PDO. 

When however the datatype is ARRAY or RECORD as above then subindex 0 holds the number of items available (0 means none and 1 means one item). Starting from subindex 1 the items enumerate according to subindex 0. 

It is possible not all subindexes hold values and can be NULL (an SDOread will return -1). This is called a sparse array.

When an item has a size <8 bits you need to assemble a combination of bits to align to a multiple of 8 bits. You can either use index 0x0000:00 with the appropriate size as a filler or combine enough actual PDO bit elements. Keep in mind that most slaves will not allow you to mix and match arbitrary bits from non monotonous items. In your example you most likely will have to map all bits of index 0xf200. The split in bits is just to make it easy for the application to understand the meaning of your mapping.

But all this is well explained in the EtherCAT protocol documentation.

farawayfromhome, 2019-05-17:
For example, here is the PDO information I need to map.
[image]
It says 0x60FE:00 do not need to PDO map. So I suppose to map 0x60FE:01 and 0x60FE:02. And my code is:
` 
retval += ec_SDOwrite(1, 0x1C12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
retval += ec_SDOwrite(1, 0x1C13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

u8val = 0;
retval += ec_SDOwrite(1, 0x1600, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

u32val = 0x609A0020;
retval += ec_SDOwrite(1, 0x1600, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	
u32val = 0x60fe0120;
retval += ec_SDOwrite(1, 0x1600, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);

u8val = 2;
retval += ec_SDOwrite(1, 0x1600, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

u8val = 0;
retval += ec_SDOwrite(1, 0x1A00, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

u32val = 0x60FD0020;
retval += ec_SDOwrite(1, 0x1A00, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);

u8val = 1;
retval += ec_SDOwrite(1, 0x1A00, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

u16val = 0x1600;
retval += ec_SDOwrite(1, 0x1C12, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);

u8val = 1;
retval += ec_SDOwrite(1, 0x1C12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

u16val = 0x1A00;
retval += ec_SDOwrite(1, 0x1C13, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);
u8val = 1;
retval += ec_SDOwrite(1, 0x1C13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
`
I wrote 0x0x609A0020 and 0x60fe0120 to 0x1600:01 via sdowrite, and then the following happens when printing map:
[image]
I don't know why 0x60FF:00 will print because I didn't configure 0x60FF at all.
I caught the packet through wireshark and found that the data I sent was indeed 0x60FE, and the slave also answered correctly.
[image]

nakarlsson, 2019-05-17:
You write to the same index twice, I guess 0x60ff just stick around.
It is weird that the 0x609A sticks since you write 0x60fe after.
Other from simple mistakes, as above, your correct from a SOEM perspective, so it is a slave issue rather than a SOEM issue.

Have you checked the SDO responses? Checking working counter doesn't mean the COE SDO as such was successful, you need to check for COE SDO issues. Consult the examples for usage of EcatError.

For further assistance I think you need to post a complete Wireshark.

farawayfromhome, 2019-05-17:
I do write the index number twice. Thank you very much for pointing it out. The code after I modified is:
`u32val= 0x609A0020;
retval += ec_SDOwrite(1, 0x1600, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
u32val = 0x60fe0020;
retval += ec_SDOwrite(1, 0x1600, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
u8val = 2;
retval += ec_SDOwrite(1, 0x1600, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);`

But after I changed it, the printed map still looks like this:
[image]

I transplant the function of 'ecatcheck' in simpletest.c into my code and find that there is no error message printed out.
And SDO does not print any of the following error messages:
`/** Error types */
typedef enum
{
   EC_ERR_TYPE_SDO_ERROR         = 0,
   EC_ERR_TYPE_EMERGENCY         = 1,
   EC_ERR_TYPE_PACKET_ERROR      = 3,
   EC_ERR_TYPE_SDOINFO_ERROR     = 4,
   EC_ERR_TYPE_FOE_ERROR         = 5,
   EC_ERR_TYPE_FOE_BUF2SMALL     = 6,
   EC_ERR_TYPE_FOE_PACKETNUMBER  = 7,
   EC_ERR_TYPE_SOE_ERROR         = 8,
   EC_ERR_TYPE_MBX_ERROR         = 9,
   EC_ERR_TYPE_FOE_FILE_NOTFOUND = 10
} ec_err_type;`

Here is the complete Wireshark:
[ethercat.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3190272/ethercat.zip)

farawayfromhome, 2019-05-17:
I do write the index number twice. Thank you very much for pointing it out. The code after I modified is:
u32val= 0x609A0020; 
retval += ec_SDOwrite(1, 0x1600, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM); 
u32val = 0x60fe0020; 
retval += ec_SDOwrite(1, 0x1600, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM); 
u8val = 2; 
retval += ec_SDOwrite(1, 0x1600, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

But after I changed it, the printed map still looks like this:
[image]

I transplant the function of 'ecatcheck' in simpletest.c into my code and find that there is no error message printed out.
And SDO does not print any of the following error messages:
/** Error types */ 
typedef enum { 
EC_ERR_TYPE_SDO_ERROR = 0,
 EC_ERR_TYPE_EMERGENCY = 1, 
EC_ERR_TYPE_PACKET_ERROR = 3, 
EC_ERR_TYPE_SDOINFO_ERROR = 4,
 EC_ERR_TYPE_FOE_ERROR = 5, 
EC_ERR_TYPE_FOE_BUF2SMALL = 6,
 EC_ERR_TYPE_FOE_PACKETNUMBER = 7, 
EC_ERR_TYPE_SOE_ERROR = 8, 
EC_ERR_TYPE_MBX_ERROR = 9, 
EC_ERR_TYPE_FOE_FILE_NOTFOUND = 10 
} ec_err_type;

Here is the complete Wireshark:
[ethercat.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3190277/ethercat.zip)

nakarlsson, 2019-05-17:
The wireshark looks OK except that the values you write don't stick. it have to be a slave issue of some kind and I recommend reading the manual once again, if your missing some detail or contact the slave vendor support.

You can always try COE SDO complete access , if the slave support it.

ArthurKetels, 2019-05-17:
You used `u32val = 0x60fe0020;`
This is not a valid PDO, you should use 0x60fe0120. Most likely the slave tested your PDO mapping at write time and rejected it silently. Second the data size is not correct 0x60fe:00 is uint8 so should be 0x60fe0008.

nakarlsson, 2019-05-17:
Isn't 0x60fe0020 valid if it is of variable type?

farawayfromhome, 2019-05-17:
Thank you all. My problem is solved now.
Index 0x60FE:00 has no PDO mapping, and only 0x60FE:01 and 0x60FE:02 have PDO mapping. 

Thank for your help, nakarlsson & ArthurKetels .

closed 2019-05-17 by farawayfromhome