OpenEtherCATsociety/SOEM #298: Input Configuration Error

KleveJ, 2019-05-15:
Hi everyone, I new to SOEM and I am running into an issue during the configuring stage.
Slaveinfo eth0 -map provides the following info

[image]

I have also had a look at [https://github.com/hefloryd/SOEM/commit/22d8f5d](url) but I am a little confused about the values on lines 64 through 72 for simple_test.

Any help setting this up would be appreciated. Ive included the ESI file and communication manual to provide as much info as I can.

ESI file

[P0012817-01082019.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3183762/P0012817-01082019.txt)

Drive Manual
[SANMOTION_R_3E_M0011697.pdf](https://github.com/OpenEtherCATsociety/SOEM/files/3183765/SANMOTION_R_3E_M0011697.pdf)

ArthurKetels, 2019-05-15:
According to ESI the default configuration should work. Slaveinfo can not read the mapping for one reason or the other. So please post a wireshark capture .pcap file.

KleveJ, 2019-05-15:
I tried the example that I linked above but swapped my SM2 and SM3 registers for the ones in the example and I wound up with this result after running slaveinfo.c

[image]

Here is the file from pcap from wireshark.

[RS3_pcap.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3184176/RS3_pcap.zip)

ArthurKetels, 2019-05-15:
I think you have to read from 5-6 onward in the manual. The manual is very precise in how the master should engage with the drive. Just follow the steps. When that works see how much you can change before the drive starts complaining.

KleveJ, 2019-05-15:
Ive read through most of it but it doesnt really tell me what I need to change to get into Safe-OP or if it does I dont know what to look for.

ArthurKetels, 2019-05-16:
Looks like you want to bake cake and don't know how to read the recipe. 

I give you two examples from 5-8

Is ESM in "Pre-Operational" state? R 0x0130 - : AL Status 0x0002)
Register read : here you (or SOEM) has to read register 0x0130 of the slave  to check pre-op state. You use ec_FPRD().

･ Sets PDO Object Index to be assigned in SM2 W 0x1C12.0=0x01 : SM2 RxPDO Entry
SDO write : here you (or SOEM) has to write SDO object 0x1c12:00 with value 0x01. You use ec_SDOwrite().

How much cleared do you want it? And if you do not understand the basic principles of EtherCAT first read up on it. There is lots of documentation available on www.ethercat,org.

KleveJ, 2019-05-16:
Thank you. Ive read what is available on the public facing part of ETC and ive applied for membership so I can get access to the knowledgebase and forums there but I havnt heard back yet.

ArthurKetels, 2019-05-16:
Hmm, trying to get a complex slave configuration done without the support of the member documentation is hard indeed. SOEM assumes you have proper knowledge of the inner EtherCAT workings. It is a lib, not a complete packaged solution.

KleveJ, 2019-05-20:
I agree with you there. I understand the basics but Im probably missing a few key details here and there which doesnt help. Ive been able to get into the OP state consistently but Im running into a issue where when I write the setup for the slave I get a watchdog configuration error. Immediately after I commented out the setup related code and it goes into OP mode as expected. Running it a second time results in an inconsistent settings error. 

so the order goes some like this"
1) Run with modified slave setup --> output Watchdog Configuration error
2) Run without modified slave setup --> output OP state Process data as normal
3) Run again without modified slave setup --> out inconsistent settings error
4) see step 1

The modified file looks like this:
`/** \file
 * \brief Example code for Simple Open EtherCAT master
 *
 * Usage : simple_test [ifname1]
 * ifname is NIC interface, f.e. eth0
 *
 * This is a minimal test.
 *
 * (c)Arthur Ketels 2010 - 2011
 */

#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "ethercat.h"

#define EC_TIMEOUTMON 500

char IOmap[4096];
OSAL_THREAD_HANDLE thread1;
int expectedWKC;
boolean needlf;
volatile int wkc;
boolean inOP;
uint8 currentgroup = 0;

static int RS3_write8(uint16 slave, uint16 index, uint8 subindex, uint8 value)
{
    int wkc;
    wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
    return wkc;
}

static int RS3_write16(uint16 slave, uint16 index, uint8 subindex, uint16 value)
{
    int wkc;
    wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
    return wkc;
}

static int RS3_write8(uint16 slave, uint16 index, uint8 subindex, uint32 value)
{
    int wkc;
    wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
    return wkc;
}

static int RS3_setup(uint16 slave)
{
    int wkc;
    printf("RS3 drive setup\n");

    wkc += RS3_write8(slave, 0x1C12, 0, 0);
    wkc += RS3_write8(slave, 0x1C13, 0 , 0);
    
    wkc += RS3_write8(slave, 0x1702, 0 , 0);
    wkc += RS3_write32(slave, 0x1702, 1, 0x607A0020);
    wkc += RS3_write32(slave, 0x1702, 2, 0x60600000);
    wkc += RS3_write32(slave, 0x1702, 3, 0x00000008);
    wkc += RS3_write32(slave, 0x1702, 4, 0x60400010);
[... 240 lines cut ...]
        }
        osal_usleep(10000);
    }
}

int main(int argc, char *argv[])
{
   printf("SOEM (Simple Open EtherCAT Master)\nSimple test\n");

   if (argc > 1)
   {
      /* create thread to handle slave error handling in OP */
//      pthread_create( &thread1, NULL, (void *) &ecatcheck, (void*) &ctime);
      osal_thread_create(&thread1, 128000, &ecatcheck, (void*) &ctime);
      /* start cyclic part */
      simpletest(argv[1]);
   }
   else
   {
      printf("Usage: simple_test ifname1\nifname = eth0 for example\n");
   }

   printf("End program\n");
   return (0);
}`

nakarlsson, 2019-05-21:
I think you've copy'n'pasted something not applicable for your use-case? 

//Explicitly set flags that are (probably) invalid in EEPROM
ec_slave[slave].SM[2].SMflags = 0x10024'

KleveJ, 2019-05-21:
Thanks, that did fix the issue

closed 2019-05-21 by KleveJ