OpenEtherCATsociety/SOEM #299: EtherCAT Master License

Wanga, 2019-05-18:
> The EtherCAT Technology, the trade name and logo "EtherCAT" are the intellectual
> property of, and protected by Beckhoff Automation GmbH. You can use SOEM for the
> sole purpose of creating, using and/or selling or otherwise distributing an
> EtherCAT network master provided that an EtherCAT Master License is obtained
> from Beckhoff Automation GmbH.
> 
> In case you did not receive a copy of the EtherCAT Master License along with
> SOEM write to Beckhoff Automation GmbH, Eiserstrasse 5, D-33415 Verl, Germany
> (www.beckhoff.com).

[The license file](https://github.com/OpenEtherCATsociety/SOEM/blob/master/LICENSE) in SOEM mentioned we need the master license provided by Beckhoff, which I couldn't find in the SOEM repository.  I assume this is _"A licence for implementing an EtherCAT master which is free of charge"_ in [ETG F&Q 3.3 ](https://www.ethercat.org/en/faq.html#789). Maybe something like [this](https://bit.ly/30r2JXH)? 

Is that possible to have this file a SOEM repository? Or every users are supposed to contact Beckhoff individually?

nakarlsson, 2019-05-18:
SOEM is an EtherCAT Master library, you as an user implement the EtherCAT Master. The master license is between the EtherCAT Master vendor and Beckhoff. Hence, it can't be included in the repository.

nakarlsson, 2019-05-21:
Any further question?

Wanga, 2019-05-21:
No, thank you.

closed 2019-05-21 by Wanga