OpenEtherCATsociety/SOEM #300: After configure PDO, the Ethercat state mechine cannot switch to OP.

Jiongyu, 2019-05-19:
Dear sir,
 I am a novice in ethercat.
when I try to run demo "simple_test",the state machine can switch to the op mode.
it generate:
```
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp4s0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 20 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
Processdata cycle    1, WKC 3 , O: 00 00 00 00 00 00 00 00 I: 00 00 00 00
Processdata cycle    2, WKC 3 , O: 00 00 00 00 00 00 00 00 I: 00 00 00 00 
Processdata cycle    3, WKC 3 , O: 00 00 00 00 00 00 00 00 I: 00 00 00 00 
.........
```

but I try to map the pdo by sdo,the state machine can not to switch to op mode still in the safe-op mode together with error.

Please let me know what's the cause of this error and how to rectify this.
thank you!!!

my ethercat init code:

```
bool EtherCatManager::initSoem(const std::string& ifname) {
  // Copy string contents because SOEM library doesn't
  // practice const correctness
  const static unsigned MAX_BUFF_SIZE = 1024;
  char buffer[MAX_BUFF_SIZE];
  size_t name_size = ifname_.size();
  if (name_size > sizeof(buffer) - 1)
  {
    fprintf(stderr, "Ifname %s exceeds maximum size of %u bytes\n", ifname_.c_str(), MAX_BUFF_SIZE);
    return false;
  }
  std::strncpy(buffer, ifname_.c_str(), MAX_BUFF_SIZE);

  printf("Initializing etherCAT master\n");

  if (!ec_init(buffer))
  {
    fprintf(stderr, "Could not initialize ethercat driver\n");
    return false;
  }

  /* find and auto-config slaves */
  if (ec_config_init(FALSE) <= 0)
  {
    fprintf(stderr, "No slaves found on %s\n", ifname_.c_str());
    return false;
  }

  printf("SOEM found and configured %d slaves\n", ec_slavecount);
  for( int cnt = 1 ; cnt <= ec_slavecount ; cnt++)
  {
    // MINAS-A5B Serial Man = 066Fh, ID = [5/D]****[0/4/8][0-F]*
    printf(" Man: %8.8x ID: %8.8x Rev: %8.8x %s\n", (int)ec_slave[cnt].eep_man, (int)ec_slave[cnt].eep_id, (int)ec_slave[cnt].eep_rev, IF_MINAS(ec_slave[cnt])?" MINAS Drivers":"");
    if(IF_MINAS(ec_slave[cnt])) {
      num_clients_++;
    }
  }
  printf("Found %d MINAS Drivers\n", num_clients_);

  if (ec_statecheck(0, EC_STATE_PRE_OP, EC_TIMEOUTSTATE*4) != EC_STATE_PRE_OP)
    {
[... 228 lines cut ...]

  if(ec_statecheck(0,EC_STATE_OPERATIONAL, EC_TIMEOUTSTATE) != EC_STATE_OPERATIONAL)
  {
    fprintf(stderr, "OPERATIONAL state not set, exiting\n");
    return false;
  }

  ec_readstate();
  for( int cnt = 1 ; cnt <= ec_slavecount ; cnt++)
    {
      if (! IF_MINAS(ec_slave[cnt])) continue;
      printf("\nSlave:%d\n Name:%s\n Output size: %dbits\n Input size: %dbits\n State: %d\n Delay: %d[ns]\n Has DC: %d\n",
	     cnt, ec_slave[cnt].name, ec_slave[cnt].Obits, ec_slave[cnt].Ibits,
	     ec_slave[cnt].state, ec_slave[cnt].pdelay, ec_slave[cnt].hasdc);
      if (ec_slave[cnt].hasdc) printf(" DCParentport:%d\n", ec_slave[cnt].parentport);
      printf(" Activeports:%d.%d.%d.%d\n", (ec_slave[cnt].activeports & 0x01) > 0 ,
	     (ec_slave[cnt].activeports & 0x02) > 0 ,
	     (ec_slave[cnt].activeports & 0x04) > 0 ,
	     (ec_slave[cnt].activeports & 0x08) > 0 );
      printf(" Configured address: %4.4x\n", ec_slave[cnt].configadr);
    }
  printf("\nFinished configuration successfully\n");
  return true;
}
```

ArthurKetels, 2019-05-19:
I can not download the Elmo xml. But the wireshark trace shows that the slave does not accept SDO download (is SDOwrite) to objects 16xx and 1axx. Object 1cxx download is fine however. 

Just a guess, but some slave with CompleteAccess CoE functionality do not accept PDO mapping modifications one by one. There you must use CA to change all entries of the object in one go. You could test this.

Jiongyu, 2019-05-20:
@ArthurKetels , hello , thanks for your time.

there is new elmo xml download:
[Elmo ECAT DriveGcon_v38_B03.xml](https://drive.google.com/uc?authuser=0&id=1VKE8YTHcbP2i7b6jzU0V2hAvlQ5jKQSi&export=download)

there  is elmo cia301:
[elmocia301](https://drive.google.com/uc?authuser=0&id=1_t3fQGrQ2N1-IbxEmniOlaQkttPm8xbz&export=download)

pardon me, what is CA?
how to change all entries of the object in one go?
could you give same example?

thank you.

ArthurKetels, 2019-05-20:
Thanks for the repost of the ESI file. This file shows that objects 16xx and 1axx are fixed. Meaning that you can not change them. It is possible however to select a number of TxPDO and RxPDO to create your own mapping. Also take into account that some PDO restrict other PDO.

Example from your ESI below. RxPDO 0x1600 can not be changed, is attached to SyncManager 2 and can not be mapped together with RxPDO 0x1601 0x1602 0x1603 0x1604 0x1605 or 0x1606. Tying to do so will result in a fail to go to OP mode.

				<RxPdo Fixed="1" Sm="2">
					<Index>#x1600</Index>
					<Name>Outputs</Name>
					<Exclude>#x1601</Exclude>
					<Exclude>#x1602</Exclude>
					<Exclude>#x1603</Exclude>
					<Exclude>#x1604</Exclude>
					<Exclude>#x1605</Exclude>
					<Exclude>#x1606</Exclude>
					<Entry>
						<Index>#x607A</Index>
						<SubIndex>0</SubIndex>
						<BitLen>32</BitLen>
						<Name>Target Position</Name>
						<DataType>DINT</DataType>
					</Entry>
					<Entry>
						<Index>#x60FE</Index>
						<SubIndex>1</SubIndex>
						<BitLen>32</BitLen>
						<Name>Digital Outputs</Name>
						<DataType>DINT</DataType>
					</Entry>
					<Entry>
						<Index>#x6040</Index>
						<SubIndex>0</SubIndex>
						<BitLen>16</BitLen>
						<Name>Control word</Name>
						<DataType>UINT</DataType>
					</Entry>
				</RxPdo>

Jiongyu, 2019-05-20:
@ArthurKetels , hello , thank for you help, I try to not change the pdo map and just use the defalut map , but there are still cannot switch to op mode.
there are wireshark file:
[cannot_go_to_op_mode2.pcapng](https://drive.google.com/uc?authuser=0&id=18AH1wMZyk-QdVGoiaZ0LqflV_vudnkQq&export=download)

thank you.

ArthurKetels, 2019-05-20:
There is nothing wrong this time with the master implementation. The problem is in the slave configuration. But you also made an very very complex mapping. So I would suggest to first try a simple mapping. Just use the factory default and change one thing. Then go from there.

Jiongyu, 2019-05-20:
@ArthurKetels , I find that 0x1c12 map pdo are exceed 5 and the state machine can not switch to op mode.
after that, the ethercat init success. but I got new error:
`error:channel(224) is larget thatn Input bits (224)`

wireshark:
[error:channel](https://drive.google.com/uc?authuser=0&id=1BeL0gkpAE4pOjNZN52FznHtXXxXxSk7i&export=download)
  thank you.

nakarlsson, 2019-08-10:
still an issue?

nakarlsson, 2020-03-03:
@Jiongyu , can we close this issue?

closed 2020-03-08 by nakarlsson