OpenEtherCATsociety/SOEM #301: Selecting PDO index

robotixdeveloper, 2019-05-20:
I have a simple question. My slaves have two receiving PDOs configured (PDO1 at Index 0x1600 and PDO2 at Index 0x1601).
How can I tell ec_send_processdata() which PDO to address on the slave side?

nakarlsson, 2019-05-21:
Given both are configured to be used, check slaveinfo -map to get a feeling for how the get mapped to the IOmap. ec_send_processdata send the IOmap. You write/read via the ec_slave[x].inputs and .outputs pointers.

If you need to configure which to use, check for example, #300 , for some guidance how it is done.  I can't vouch for the data in #300 but you'll see the concept.

robotixdeveloper, 2019-05-21:
Thank you for your answer, but I still do not understand how I can switch between PDO1 or PDO2. How can I send out_PDO1 and out_PDO2?

typedef struct PACKED { int16 output1; int16 output2; } out_PDO1;
out_PDO1 *out_PDO1;

typedef struct PACKED { int16 output1; int16 output2; } out_PDO2;
out_PDO2 *out_PDO2;

out_PDO1->output1 = 123;
out_PDO2->output1 = 456;

int iomap_size = ec_config_map(iomap_);
ec_send_processdata();

nakarlsson, 2019-05-21:
The structures should point to the data in the IOmap.
out_PDO1 = (out_PDO1 *)ec_slave[slave number].outputs
out_PDO2 = (out_PDO2 *)(ec_slave[slave number].outputs + sizeof(eout_PDO1)) // given the are mapped like that by the slave.
in_PDO1 = (in_PDO1 *)ec_slave[slave number].inputs

loop
send_ processdata
recv_processdata
out_PDO1->outputs1 = 123
local_in = in_PDO1->inputs1
...

robotixdeveloper, 2019-05-22:
Thanks for the input, I implement it during the next days and report feedback.

nakarlsson, 2019-08-10:
Fixed I guess

closed 2019-08-10 by nakarlsson