OpenEtherCATsociety/SOEM #302: AX5000 support startuplist reading

micFhi, 2019-05-21:
Dear all,

it seems to me that the problems in #127 with the AX5000 series from Beckhoff and the problem of loading the startuplists onto the drive on every startup have not been successfully closed or the solution is not completely documented. I saw that the [linuxcnc rep](https://github.com/sittner/linuxcnc-ethercat/tree/master/examples/initcmds ) implemented the functionality successfully, and I was thinking, if this solution could be adopted.

Let me know your thoughts. If anybody has a fully working code example for this issue, it would be very nice to have a well working solution and to close this gap. 

Thank you so much!

ArthurKetels, 2019-05-22:
This is a slave issue and has nothing to do with SOEM. Even if you get it to work using the list from linuxcnc this is only for that specific combination drive and motor. If you want support for this drive then you have to build a function that generates the proper settings depending on a set of input parameters.

micFhi, 2019-05-22:
It seems to me that [this repo](https://github.com/sittner/linuxcnc-ethercat/blob/master/src/lcec_ax5200.c)  implements that kinda function.

nakarlsson, 2020-03-09:
Can we close this?

nakarlsson, 2020-05-09:
Inactive

closed 2020-05-09 by nakarlsson

fmros, 2020-10-14:
https://github.com/OpenEtherCATsociety/SOEM/issues/447