OpenEtherCATsociety/SOEM #304: Question on proper slave initialization

jopado1, 2019-05-24:
I have slave initialization code that starts like this:
```
   if (ec_init(ifname))
   {
      printf("%s: ec_init on %s succeeded.\n", __BASEFILE__, ifname);
      /* find and auto-config slaves */

      if ( ec_config_init(FALSE) > 0 )
      {
         int mapSize = 0;
         
         printf("%s: %d slaves found and configured.\n", __BASEFILE__, ec_slavecount);

         mapSize = ec_config_map(&IOmap);
         if (mapSize==0)
         {
            printf("%s: PDO mapping failed.\n", __BASEFILE__);
            ec_close();
            return;
         }

         ec_configdc();

         printf("%s: Slaves mapped (IOmap size=%d), changing state to SAFE_OP.\n", __BASEFILE__, mapSize);
         /* wait for all slaves to reach SAFE_OP state */
         ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);
         
         /* Output some info on slaves */
         for(idx = 1; i<=ec_slavecount ; idx++)
         {
            /* Output some relevant info on slaves */
	    printf("Slave %d Name: %s. Slave info follows:\n", idx, ec_slave[idx].name);
            printf("  State=0x%2.2x StatusCode=0x%4.4x : %s\n",
                   ec_slave[idx].state, ec_slave[idx].ALstatuscode, ec_ALstatuscode2string(ec_slave[idx].ALstatuscode));
			
		...	
		...
		 }
```
Normally I see output like this:
    amc.c: Initializing servo drives.
    amc.c: ec_init on mac0 succeeded.
    amc.c: 2 slaves found and configured.
    amc.c: Slaves mapped (IOmap size=34), changing state to SAFE_OP.
    Slave 1 Name: AMC. Slave info follows:
       State=0x02 StatusCode=0x0000 : No error

But sometimes I see the following and the slave needs to be reset to recover. If I restart the master side sofware, this same behavior repeats.
    amc.c: Initializing servo drives.
    amc.c: ec_init on mac0 succeeded.
    amc.c: 2 slaves found and configured.
    amc.c: Slaves mapped (IOmap size=34), changing state to SAFE_OP.
    Slave 1 Name: AMC. Slave info follows:
       State=0x11 StatusCode=0x0016 : Invalid mailbox configuration

So a couple of questions:
  1) Is this init sequence correct? 
  2) Is there a way to recover the slave without power cycling it when this occurs?

Thank you

ArthurKetels, 2019-05-25:
If you could post a wireshark capture of a good an a bad start-up sequence I could tell you more. It could be as simple as a timeout that needs adjustment.

jopado1, 2019-05-30:
I do not have direct access to the equipment so I am trying to arrange for some cabling to allow for a wireshark capture.  However, I do have output from slaveinfo with the map option if that might shed any light. See attached files.  This is a servo drive made by AMC - I have a question out to them regarding if their slaves pass the CTT tool as part of their testing. Will let you know on that.
[20190529--slavinfo1.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3236993/20190529--slavinfo1.txt)
[20190529--slavinfo2.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3236994/20190529--slavinfo2.txt)

 Thank you.

ArthurKetels, 2019-05-30:
It looks like the eeprom read of the slave times out for one reason or the other. This leads to wrong mailbox size or CoE identifications. The solution might be as simple as to increase (like double) the EC_TIMEOUTEEP constant (ethercattype.h). And while at it, just double all timeouts. Then test again.

jopado1, 2019-06-03:
Doubled the timeouts and still happens sometimes.  I've attached wirshark traces for working and non-working cases. Thank you for having a look.
[PDO-mappint-nowork.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3248399/PDO-mappint-nowork.pcapng.txt)
[PDO-mappint-worked.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3248400/PDO-mappint-worked.pcapng.txt)

ArthurKetels, 2019-06-03:
Thanks for the wireshark traces. They do show the problem at hand. And it is not that obvious.

The problem originates at line 122 of your nowork capture. This is the 2nd half of the eeprom readout function that tries to figure out if the slave has a mailbox. The function ecx_eeprom_waitnotbusyAP(ecx_contextt *context, uint16 aiadr,uint16 *estat, int timeout) is called that waits for the eeprom in the slave to get out of busy state and read the data. But this function returns with a timeout (and without reading the data) within 74us. The actual timeout value is 20ms.

As a result no mailbox length is found and this slave is regarded as having no mailbox. The following state change to pre-op fails with a mailbox configuration error from the slave (correct).

I do notice that in both captures your line timestamps are very suspicious. Many occasions there are 0us between tx and rx frames. So my guess for now is that your OS clock is not continuous and/or monotonous. What is your hardware and OS for these tests? Can you do the same test on another box (preferably intel processor with linux-rt).?

If not you could also trap timeouts within the eeprom read functions and log them.

jopado1, 2019-06-05:
Its VxWorks 7 on a powerPC processor running 64-bit.  I'd be surprised if there were clock/timing issues as we are doing other time sensitive processing.  We do however run the Vx7 system clock at 20000 (60 is the default). Maybe that is contributing?  No opportunity to run this on other hardware.  I'll have a look at the timeout loops and see if they perform as expected. Will let you know. If you think of anything else to look at please let me know. Thx.

nakarlsson, 2019-06-05:
Do a systemview log, you can add the frame Idx used by the stack as a user event and which may help you see/match it the the wiresharklog. Then you’ll see the timing in detail.

jopado1, 2019-06-06:
I think your earlier point regarding clock being non-monotonic may be the key.  I put some test code together based on on gettimeofday and I do see unexpected results.  Occasionally when two consecutive calls are made to gettimeofday, the second gives a time prior to the first. Not good.  gettimeofday is not guaranteed to be monotonic. I modified osal_gettimeofday to use clock_gettime with CLOCK_MONOTONIC.  I expect this will fix this issue when I test later today. Will let you know.

nakarlsson, 2019-06-07:
If it works, please create a pull request. It make sense to change.

nakarlsson, 2019-06-07:
Looks like you still have the EEPROM fail to read issue. Very the timer functions with systemviewer?

ArthurKetels, 2019-06-08:
I looked a bit into the VxWorks nicdrv.c. This is code that I am not able to test because I do not have VxWorks. So here I rely on work of others.

But I found some fishy code in mux_rx_callback. From line 588:

            else if ((status == ERROR) && (errno == S_objLib_OBJ_UNAVAILABLE))
            {
                /* Try to empty the MSGQ since we for some strange reason
                * already have a frame in the MsqQ,
                * is it due to timeout when receiving?
                * We want the latest received frame in the buffer
                */
                port->pktDev.overrun_count++;
                NIC_LOGMSG("mux_rx_callback: idx %d MsgQ overrun\n", idxf,
                    2, 3, 4, 5, 6);
                M_BLK_ID trash_can;
                if (msgQReceive(msgQId, (char *)&trash_can,
                    sizeof(M_BLK_ID), NO_WAIT) == OK)
                {
                    /* Free resources */
                    netMblkClChainFree(trash_can);
                }

                status = msgQSend(msgQId, (char *)&pMblk, sizeof(M_BLK_ID),
                    NO_WAIT, MSG_PRI_NORMAL);
                if (status == OK)
                {
                    NIC_WVEVENT(ECAT_RECV_RETRY_OK, (char *)&length, sizeof(length));
                    ret = TRUE;
                }
}

If this hits then it would explain lost information in SOEM. You can not simply discard received packets without penalty. 

I am also not sure if the message queue is the right solution to pass packets from the call-back to SOEM. For sure it will cause a lot of scheduling overhead. How are time-outs handled (always via system ticks?) and how about priority inversions?  

```
   static int ecx_recvpkt(ecx_portt *port, int idx, int stacknumber, M_BLK_ID * pMblk, int timeout)
   {
      int bytesrx = 0;
      MSG_Q_ID  msgQId;
      int tick_timeout = max((timeout / 1000), 1);
```
The tick_timeout calculation is obviously wrong when the system tick rate is not 1000Hz.

nakarlsson, 2019-06-08:
Thanks for the driver review, 

It is designed to not poll for packets and the callback mux_rx_callback is called from the NIC ISR task passning a ref to a received mBlk frame.
So, yes one extra scheduling but no need for mutex locks in receive and send. So, yes the time resolution is systemtick. But, less busy/while looping.

If the code in 588 hits the packet(one Idx) has been abandoned/timed-out, every frame Idx has its own mbox with space for one frame ref( mBlk pointer) at the time, pretty much as all other drivers having a buf_t per Idx.

Of course using OS primitives limit the timeouts to system tick, per design. I see no reason for priority inversion. The user send a frame with Idx and waits for it to return, becomes a consumer. The ISR receive the frame and become the producer posting a frame ref back.

BUG, good catch, should of course use the proper system tick.

NOTE, it is simple to turn the driver to a polling using the mux polling API.

ArthurKetels, 2019-06-09:
I am not against using the systemtick or the message passing system, it is simple and robust. I am more worried about performance. I guess VxWorks users want something better than windows performance. 

When the system tick is at 1 kHz it is hard to recover quickly from lost frames. Suppose the PDO are exchanged at 1 kHz rate and there is a communication error. a sensible timeout will leave enough time to retry and still be within the 1ms. With the current code this is impossible.

Then there are some internals in VxWorks that I do not understand. What is the context of the NIC receive callback? What priority does it run? Can it be called again while still processing an earlier frame?  Do messagequeues transmits always require a reschedule? 

Did you ever come to measuring the added latency in the nicdrv?

nakarlsson, 2019-06-09:
This driver is application specific; it is designed for a NIC running 2 Layer 2 protocols in parallel using VLAN, EtherCAT being one of the protocols. The other is to support a legacy system running a custom Layers 2. 
I'll consult my customer if we can remove the VxWorks port from this repo and keep it internal 

First question:
It is possible to use receive_processdata(0)
``
send_pdata()
do_stuff()
/* The frame is expected to be in */
while( receive_pdata(0) != expected_WCK && retrys++ < max_retrys)
   /* Lost frame, send again */
   send_pdata()
``

Second question:
The NIC receive callback handle all NIC interrupts from a system task tNetX, where you in new systems have a tNetX per NIC. In legacy system there were only one tNet0. It is configurable on what priority it run and it typical look like this.

NIC RECV JOB(frame)
for(;number_protocols;++)
   if(frame_type)
     if(call_recv_hook(frame, frame_type) == handled)
        break;
 
It is possible to run in polling mode but that also introduce penalty since it must copy the frame on receive while the NIC task can make use of frames assigned HW DMA. It is never black or white.

ArthurKetels, 2019-06-09:
As long as someone can still test the code for VxWorks I am fine with leaving it in the repo. Just want to avoid dead code. It is not like the windows target is that performant, and it is still in.

nakarlsson, 2019-08-10:
https://github.com/OpenEtherCATsociety/SOEM/releases/tag/v1.4.0 have been released having some VxWorks improvments mentioned above

nakarlsson, 2020-03-03:
@jopado1, re-open if needed. Improvements have been added

closed 2020-03-03 by nakarlsson