OpenEtherCATsociety/SOEM #307: Question about configuring a slave

darrabito, 2019-06-07:
Hello all,
I am very new to EtherCAT so I'm sorry if this may seem trivial.

I had been working on trying to configure an EL2535 and it would not enter OP mode. I saw a similar issue where it was recommended to disable CA so I did and that worked. I am able to configure the RW data with some SDOwrites and verify it by running slaveinfo -sdo.
My question is how I would enable a RO entry to enable the current controller on this device.
I try to configure it the same way as the the RW data but nothing happens.

Here is the documentation for the EL2535. The object descriptions are on page 110.
EL2535 ESI: 
[Beckhoff EL25xx.xml.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3266524/Beckhoff.EL25xx.xml.txt)

nakarlsson, 2019-06-07:
It is part of the processdata, you van write it as first byte bit 5 in the IOmap, read examples and tutorial on howto read/write processdata.

[0x0000.5] 0x7000:0x06 0x01 BOOLEAN      Enable <- logic address byte 0 bit 5, correspond to IOmap[0] bit 5

darrabito, 2019-06-07:
So do I want to do this?

ec_slave[1].outputs[0] = 5;

assuming slave 1 is what I want. Where would I find some good examples on processdata? The ethercat  site itself? I've been reading Beckhoffs documentation and that doesn't seem nearly sufficient enough.

darrabito, 2019-06-07:
Thank you, that was all i needed to know, I got it working

closed 2019-06-07 by darrabito