OpenEtherCATsociety/SOEM #308: Slaves lost in operational state

farawayfromhome, 2019-06-10:
Hi.
I've been fighting with the simple_test example.All the slaves can reach to OP, but when several cycles of process data are sent, all slaves are lost, which is a must. I don't know what caused it. Whether there are replies from the slaves station but the master does not receive them or the slaves do not reply at all.
[image]
[ecat_capture.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3271622/ecat_capture.zip)

ArthurKetels, 2019-06-10:
All slaves are and remain in OP. All packets look ok. For some reason the simple_test loop does no longer see a correct working counter (=18). The recovery routine kicks in and restores all slaves (no need but anyway). This loop keeps running because the wkc remains not ok.

You could print the actual wkc to see what is being returned.

What OS and cpu are you using?

farawayfromhome, 2019-06-11:
Thank you for your reply.
I did as you say to print the the actual wkc out.And I found that wkc suddenly became -1.
I use RTEMS 4.11 and Zynq7020.

[image]

ArthurKetels, 2019-06-11:
WKC -1 means that no valid frame was received. And from the wireshark trace we can see all frames have returned normal. So you will have to instrument nicdrv.c with debug info to trace down the issue. The routine in question is : int ecx_inframe(ecx_portt *port, int idx, int stacknumber)

farawayfromhome, 2019-06-11:
Thanks.
As you suggested, I found that the program could not run into the ether isr_handler in freebsd when the data was not received. I will continue to trace this problem.

I have another problem, after I used ec_slave[slave_num].blockLRW = 1
The code:
expectedWKC = (ec_group[0].outputsWKC * 2) + ec_group[0].inputsWKC; 
should be changed to :
expectedWKC = ec_group[0].inputsWKC  ?

farawayfromhome, 2019-07-04:
Problem solved

closed 2019-07-04 by farawayfromhome