OpenEtherCATsociety/SOEM #311: Data cannot be transferred or stored to the application because of local control

darrabito, 2019-06-12:
Hello,

I am trying to do a SDOwrite during pre-op to 0x1c13:01 with a value of 0x1a01 but I get  error:08000021 Data cannot be transferred or stored to the application because of local control.
Would someone be able to explain what exactly this error means?

If you need any additional information, let me know.

nakarlsson, 2019-06-13:
You might need to write 0 du 0x1c13:0 first, update 0x1c13:x , fnalize with 0x1c13:0 = number of entries in 0x1c13

You can find info in ETG 1020 PDO assign

darrabito, 2019-06-13:
Thank you, that fixed my issue, I am working on getting a membership to access those documents now.

closed 2019-06-13 by darrabito

stevedanomodolor, 2020-07-15:
How did you fix the issue? please I am having the same problem

stevedanomodolor, 2020-07-15:
@nakarlsson  can you ellaborate more on the answer, I dont understand what you mean