OpenEtherCATsociety/SOEM #313: Serial number from EEPROM

robert-koenig, 2019-06-14:
Hello, 

I am new to GitHub and SOEM also, so I don't know if this is the right way to propose a change. 

I noticed that by configuring a slave the serial number of the device is not read from the EEPROM. It should be easy to do it by:
- adding eep_sn after eep_rev in struct ec_slave in ethercatmain.h
- adding ECT_SII_SN = 0x000e to enum of item offsets in SII general section in ethercattype.h 
- extend ecx_config_init() by reading context->slavelist[slave].eep_sn

In my local copy of SOEM this works. Is there a certain reason why S/N is not read?

Please tell me if there is a better way to make a proposal like this.

Regards,
Robert

nakarlsson, 2019-06-15:
Do a pullreques from you own fork, Google on GitHub an pullrequest and fork.
Trivial and single purpose commits

ArthurKetels, 2019-06-15:
I am already preparing a pull req with many things. This is one of them.

mwh-newtec, 2020-12-09:
@ArthurKetels 
Any update on this pull req?
Can't seem to find it in the project.
Also have a "unofficial" commit like the add that robert-koenig suggests.
Would be nice to clean it up if it's something that you have implemented already.

Best regards Mathias