OpenEtherCATsociety/SOEM #314: ec_config_map(&IOmap) causes Error and how to change IOmap values

miscz, 2019-06-19:
Hello!

I already mentioned some earlier problems in issue #261. After explaining the vendor my problem, they provided a new eeprom, which I could successfully upload with the eepromtool and the communication seems to work fine now. Slaveinfo gives me the following information:

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s31f6 succeeded.
1 slaves found and configured.
Calculated workcounter 1

Slave:1
 Name:? M:000000e4 I:264860d3
 Output size: 0bits
 Input size: 64bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000000e4 ID: 264860d3 Rev: 00020000
 SM0 A:1800 L:  16 F:00010026 Type:1
 SM1 A:1900 L:  16 F:00010022 Type:2
 SM2 A:1600 L:   0 F:00000064 Type:3
 SM3 A:1a00 L:   8 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:1a00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 16 rd: 16 MBX protocols : 04
 CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```

However I still get error messages on the controller, saying the rpdo length is not valid. When trying the red_test example, I noticed this error does not occur. Apparently this problem is due to the `ec_config_map(&IOmap)` call. Since red_test does not use it, the program runs without any errors and also the dc seems to work properly . So my questions are:

- Do I really need `ec_config_map(&IOmap)`?
- When does the mapping happen in red_test? Is it in `ec_config(FALSE, &IOmap)`?

Inside the cyclic loop of red_test I tried to read and write the IOmap values by simply addressing the elements such as `IOmap[6]= 1;`.

- Is this the correct way to change values?

I connected another computer to the controller, where I can read the actual PDO values. When changing the values with `IOmap[6]= 1` it takes about 1 second, such that the new value is also changed on the controller. In this one second the command `IOmap[6]= 1` was permanently send inside the cyclic loop.

- Shouldn't the new value occur right after one cycle is finished?

Sorry for this bunch of questions, but I hope you guys can help me with this issues.

Michael

nakarlsson, 2019-06-28:
If you study the functions you'll find out how they are related. Basically you need to config_init to setup mailboxes, you need to config_map to read process data and create the SOEM processdata mapping. All this is shown in the examples.

Second, the slave report 0bits outputs, yet you try to write datat to it. Something isn't correct here. Please all relevant information here, wrireshark, ESI, ObjectDictionary dump with slaveinfo?

miscz, 2019-07-02:
Thank you for your answer. I made it working now. I had to add a hook to set the subindex 0 of 0x1C12 and 0x1C13 to 1. Now `slaveinfo` gives me:

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s31f6 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:? M:000000e4 I:264860d3
 Output size: 64bits
 Input size: 64bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000000e4 ID: 264860d3 Rev: 00020000
 SM0 A:1800 L:  16 F:00010026 Type:1
 SM1 A:1900 L:  16 F:00010022 Type:2
 SM2 A:1600 L:   8 F:00010064 Type:3
 SM3 A:1a00 L:   8 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:1600 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000008 Ll:   8 Lsb:0 Leb:7 Ps:1a00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 16 rd: 16 MBX protocols : 04
 CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```

This should be right now, right? `simple_test` works also fine and I can display the output and input values. If you still want to see the wireshark, ESI, ObjectDictionary dump let me know.

Unfortunately one question is still left: How do change output values in the mapping? I can see all input values that the slave is sending, but I do not know how to send values from the master to the slave? Like mentioned in the last post `IOmap[6]= 1` does not work accurately. I did not find any examples where I can see how to do this.

nakarlsson, 2019-07-04:
IOmap[6]= 1 should work if all things are OK, (slave state, IO mapping) etc..
But I suggest you read the documentaion -> "Accessing data through IOmap" or search older "issues" for more information.

https://openethercatsociety.github.io/doc/soem/index.html

miscz, 2019-07-04:
I already read the documentation before and I was wondering how data access is solved there. Like mentioned in the first sentence "IOmap is the fastest mechanism for accessing slaves' IO data" I was expecting to see something like `IOmap[6]= 1`. If I understood it right, a new function `set_output_int16 ` was created, to access the output data. In the code I did not find a line where `IOmap[..]= ..` is used.

Just for clarification: I should write my own function `set_output_int16()` in order to change values in the slave? Or is `IOmap[6]=1` doing the exact same like `set_output_int16(EL4001_1,0,slave_EL4001_1.out1)` with the same efficiency and I can use them equivalently?

nakarlsson, 2019-07-04:
You can use IOmap[6]=1 if your comfortable with that, in the end it is all there is about, reading and writing the IOmap. How you reference data in the IOmap is up to you to know,

nakarlsson, 2019-07-29:
Answer provided

closed 2019-07-29 by nakarlsson