OpenEtherCATsociety/SOEM #315: DC Configuration Cause Failing to Reach OP State

farawayfromhome, 2019-06-28:
I am testing a slave. I found that if I configure the DC first and then call ec_config_map(&IOmap), just like this:

ec_configdc();
ec_slave[1].PO2SOconfig = DC_Config;
ec_config_map(&IOmap);

int DC_Config(uint16 slave)
{
    ec_dcsync0(slave, TRUE, 2000000U, 0);
}

the slave cannot reach the OP state. But I read the value of 0x1C32:1 is 2, indicating that it is SYNC mode. And no error code print out.
But if the sequence is : 

ec_config_map(&IOmap);
ec_configdc();
ec_dcsync0(slave, TRUE, 2000000U, 0);

then the slave can reach the OP state, but I read the value of 1C32:1 is 1.

Here is information for failing to reach OP state:
[image]

farawayfromhome, 2019-06-28:
wireshark capture
[Can Not Reach OP State.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3338400/Can.Not.Reach.OP.State.zip)

nakarlsson, 2019-06-28:
Run for a while loonger (default 10s) and the slave should timeout the state change to OP and report an error.

farawayfromhome, 2019-06-28:
As you said, there is an error.
Is it due to my configuration order? Should I configure DC in the second sequence?

[image]

nakarlsson, 2019-06-28:
Your not sending process data synchronous to the DC clock, you can search old issues on Distributed Clocks or check red test for example and discussions on the subject.

farawayfromhome, 2019-07-01:
I do check #142.But I do not really get it.Do you have an example of sending process data synchronous to DC when SafeOp to Op?

farawayfromhome, 2019-07-01:
I do check #142.But I do not really get it.

Do you have an example of sending process data synchronous to DC when SafeOp to Op?

nakarlsson, 2019-07-01:
You have a controller task that will control the state machine, you have a process data task that send process data periodically in parallel. 
As in the example and redtest you get feedback of the cycle time you send with, that feedback is used to calc the coming process data TX. You can check the diff of that time and some other registers in the ESC to verify that your in sync. Check the ET1100 charters in part I och DC startup. Based on that you choose when to enter OP.

In the end it boils down to;  you’re  expected the send minimum one process data per DC cycle that the slave expects.

DC comes in flavours, you can read ET1020 protocol enhancements for more info

farawayfromhome, 2019-07-02:
Thanks for your reply. It really make me understand synchronization mode more clearly .

Can I send process data in half of the synchronization cycle?

And I found that redtest is the process data sent in OP state in synchronization with DC clock. 
But in the process of SafeOp to OP, do i also need to send synchronized process data, just like in OP state?

nakarlsson, 2019-07-02:
You should send process data ahead of DC time, eg.  so, it is in place and the slave can act on it before the DC clock interrupt occur. On the DC interrupt the slave take actions if no data was present.

In general, you should send process data and become synchronous before entering OP, so yes in the process of running SafeOp to Op you should send process data. 

But be aware of that it is the slave that implements the DC checks and will be the one that set/get into ERROR.  So, read the slave ref manual on DC to understand any special requirements. Don't blame SOEM. There are lots of issues here on DC and flavors howto config/run DC.

closed 2019-07-02 by farawayfromhome