OpenEtherCATsociety/SOEM #316: Cannot find or open the PDB file error.

rasimckc, 2019-06-29:
Hi,
While running my 'VisualStudio 2017' I am getting "C:\Windows\SysWOW64\wpcap.dll.  Cannot find or open the PDB file error. and C:\Windows\SysWOW64\Pasket.dll.  Cannot find or open the PDB file error."I'm also a user of windows10 64-bit

                                                   DEBUG OUTPUT
'simple_test.exe' (Win32): Loaded 'C:\Users\HET-332400RC\CMakeBuilds\820837c8-48bf-1f31-abf9-d29c81f3f866\install\x86-Debug\bin\simple_test.exe'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\ntdll.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\kernel32.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\KernelBase.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\ws2_32.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\rpcrt4.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\wpcap.dll'. Cannot find or open the PDB file.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\sspicli.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\cryptbase.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\bcryptprimitives.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\sechost.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\ucrtbased.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\winmm.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\vcruntime140d.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\Packet.dll'. Cannot find or open the PDB file.
'simple_test.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\Packet.dll'
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\msvcrt.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\winmmbase.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\winmmbase.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\winmmbase.dll'. Symbols loaded.
'simple_test.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\winmmbase.dll'
'simple_test.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\winmmbase.dll'
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\cfgmgr32.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\ucrtbase.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\Packet.dll'. Cannot find or open the PDB file.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\advapi32.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\version.dll'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\IPHLPAPI.DLL'. Symbols loaded.
'simple_test.exe' (Win32): Loaded 'C:\Windows\SysWOW64\kernel.appcore.dll'. Symbols loaded.
The thread 0x1484 has exited with code 0 (0x0).
The thread 0x2118 has exited with code 0 (0x0).
The thread 0x1cfc has exited with code 0 (0x0).
The program '[13052] simple_test.exe' has exited with code 0 (0x0).

 I get this error then  I found wpcap.dll,Packet.dll in the C:\Windows\SysWOW64 folder.
What is this problem and how can I solve it? Is there anyone who can help me?
Thank you

nakarlsson, 2019-06-29:
Can’t you run without? If you run the EXE what does the output console say?

nakarlsson, 2019-07-01:
Take an hour or more to browse the information found in the testcode and tutorial please.
Tutorial: https://openethercatsociety.github.io/doc/soem/tutorial_8txt.html

Test code: https://github.com/OpenEtherCATsociety/SOEM/tree/master/test

nakarlsson, 2019-07-29:
answer provided

closed 2019-07-29 by nakarlsson