OpenEtherCATsociety/SOEM #317: Switch on disabled when reading SDO and PDO in parallel

wongkaiweng, 2019-07-01:
I have two Elmo Gold DC Whistles and I can successfully control both of them with PDOs. My problem came when I tried to read SDOs (in a thread) alongside with my PDO processing. The statusword switches to 0x250 (switch on disabled) when I read any SDO. Has anyone run into the same issue? I can get the SDO value but the change of the statusword affects my motor motion. 

I am running SOEM on Ubuntu 18.04. 
Thanks.

nakarlsson, 2019-07-02:
Best guess is that you run in some synchronous mode and reading SDOs in parallel with PDOs interfers with that? Does the slave remain in state OP?

wongkaiweng, 2019-07-02:
Yes, I am running in the synchronous velocity mode. I am reading SDOs and processing PDOs in parallel. The current reason for using SDO is I want to read the Emergency Events object of each drive. I know I can do `(char*)ec_elist2string()` for errors. However, I don't want to post-process the error string to figure out the slave in error so I thought doing it with SDO makes sense. I am also looking for suggestions or improvements in my approach. 

I foresee using SDO for other functionalities too so it will be good to figure out why it doesn't work. 
Most of the time it is in OP (State 8). However, it does sometimes go into SAFEOP + ERROR (14 - 1b [Sync manager watchdog]). I followed the `simple_test.c` example. Is `ec_dcsync0` something necessary? I have already done `ec_configdc()`. Thanks!

wongkaiweng, 2019-07-02:
If I don't run `ec_configdc()`, then my controllers always stay in OP. It's just that the statusword changes to 0x250 (Switch on Disabled) when I read a SDO.

wongkaiweng, 2019-07-03:
[Here](https://github.com/OpenEtherCATsociety/SOEM/files/3352753/sdo_read_packets_capture.zip) is a capture of some packets around SDOread with WireShark. 

In the capture, for example, in packet 1, you can see 37 12 in the datagram. (Row 0040 - the 7 and 8 entry). That's the location of my statusword. I can see that it is at 1237 for all the LRW messages. I have also captured FPRD and FPWR and I believe they come from my SDO request. I can see that the LRW and FPRD/FPRW messages are mixed together. It is interesting that I cannot see a 1250 in the packets I captured. Is mixing packets the problem?

nakarlsson, 2019-07-04:
There is a big gap in between frame No55 and No56 (or more important No52 & No 56, 2 LRW)  , 5ms. What happened there?
That would certainly make a synchronous drive unhappy. After that the word you refer to change, 37 12 becomes 37 02. I don't known the meaning(good or bad) of that but the time gap between LRW frames is strange.

wongkaiweng, 2019-07-08:
Here are a couple of things I have tried:
1.  Add mutex so that `ec_SDOxxxx` and `ec_xxx_proceedata` are mutual exclusive.
     - **Result**: It solves the problem for my laptop but it does not work for my desktop.

2. Use an RT patch kernel with Ubuntu:
     - **Result**: It improves my execution. I only get 0x250 sometimes. Here's the new  
[wireshark](https://github.com/OpenEtherCATsociety/SOEM/files/3370335/snippet_with_rt_patch.zip) with an RT kernel. The huge time gap disappeared with the patch but the problem still pops up sometimes.

nakarlsson, 2019-08-10:
Still an open issue?

wongkaiweng, 2019-08-11:
@nakarlsson I have no issue with one of my computers but the issue still exists for my other computer. I will close this issue for now since I think it is more computer dependent. If I have more findings, I will comment or reopen the issue.

closed 2019-08-11 by wongkaiweng

crown133, 2024-05-15:
Have you solved the problem eventually? I have the similar issue with you.

wongkaiweng, 2024-05-16:
@crown133 it has been a while but my end solution was to not do PDO and SDO in parallel. In my case, I was able to retrieve or update the same information through PDO. I had to create custom PDOs for it but it was the better solution for me then. Hope that helps.