OpenEtherCATsociety/SOEM #318: Problem with PDO Cycle

farawayfromhome, 2019-07-04:
When I debugged my code using RTEMS, I found that I sent the process data in a 1ms cycle, but the DC time interval actually read back was about 6ms. I used wireshark to capture data and found that the interval between each time I sent the data was really about 6ms. The same after I used simple_test.c.

[image]

[wireshark_capture.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3359526/wireshark_capture.zip)

ArthurKetels, 2019-07-04:
Looks like a granularity issue with your RTEMS scheduler. Can you test outside SOEM to see if you can make a task that runs 1ms? And what happens if you change your SOEM task to 2ms?

farawayfromhome, 2019-07-05:
It's ‘printf’ cost too much time.Now I delete printf the interval is normal.

nakarlsson, 2019-07-29:
Guess this is no issue anymore.

@farawayfromhome , please close not active Issues when they’ve been sorted out.

closed 2019-07-29 by nakarlsson