OpenEtherCATsociety/SOEM #319: Question about how to use ec_sync

farawayfromhome, 2019-07-13:
Since my OS does not support clock_nanosleep, I use a hardware Timer to time it. My operation is:
I use ec_sync to calculate toff, then set cycletime + toff to the timer's timing. 
When the timer expires, it starts to send data, which means that I use cycletime + toff as the interval between the next transmission. 
Every time I send, I flip the digital output of the master and flip the digital output of the slave with 0x60FE.
 Ideally, the Output of the master is 1, and the slave also flips to 1 in less than 1 ms. However, the actual Output of the master  is 1ms after the  slave's. I don't know where the problem is.

My code is:

static void ecatthread(rtems_task_argument arg)
{
    while(1)
   {
      set_private_timer_value(value);
      start_private_timer();
      //when the timer expires, there will be a sem to release
      semaphore_obtain(sem_id, WAITING_FOREVER, 0);
      if(inOP == TRUE)
     { 
        ec_send_processdata();
        wkc = ec_receive_processdata(EC_TIMEOUTRET);
     
        if (ec_slave[0].hasdc)
        {
	   /* calulate toff to get linux time and DC synced */
	   ec_sync(ec_DCtime, cycletime, &toff);
        }
    }
  }
}

farawayfromhome, 2019-07-13:
[image]

ArthurKetels, 2019-07-13:
Looks line the master and slave are nicely synced. But which scope trace is master and which is slave? What slave hardware do you use? What is the timing when you choose 2ms?

farawayfromhome, 2019-07-14:
The yellow one is slave’s.The DC cycle time is 1ms now.
The current situation is that the master flips the slave station to 0 before the N-1th SYNC event, and the master station flips the slave station to 1 before the Nth SYNC event. Then the slave is the Nth SYNC event flipped to 0, and the N+1th SYNC event is flipped to 1.
My understanding of synchronization is that the slave should flip to 0 in the N-1 SYNC event and the N SYNC event to 1 instead of the next SYNC event.
Is my understanding correct?

发自我的iPhone

farawayfromhome, 2019-07-14:
If the DC cycle time is 2ms ，the situation is the same.Just the interval change to 2ms.
I am using a common digital output circuit.

发自我的iPhone

ArthurKetels, 2019-07-14:
I am sorry, but I still do not have enough information about your test set-up to give an educated answer as why the reported behavior. 

One common error is that a sync0 synchronized slave should receive its updated frame well before the sync0 event. This to allow the slave to copy and prepare the data on time. When you send the frame ON the sync0 event you are already too late and the update will happen on the next event.

farawayfromhome, 2019-07-15:
My etthread has a cycletime of 1ms, dc is also set to 1ms, and no shift time is set. Then I use ec_sync to get toff.
50us is used in the formula of delta calculated in ec_sync, is it used to make data arrive at 50us before the sync0 event? If so, should I change 50us to a larger value?

farawayfromhome, 2019-07-15:
Should I change the code in ec_sync?
static void ec_sync(int64 reftime, int64 cycletime , int64 *offsettime)
{
   static int64 integral = 0;
   int64 delta;
   /* set linux sync point 50us later than DC sync, just as example */
   delta = (reftime - 50000) % cycletime;
   if(delta> (cycletime / 2)) { delta= delta - cycletime; }
   if(delta>0){ integral++; }
   if(delta<0){ integral--; }
   *offsettime = -(delta / 100) - (integral / 20);
   gl_delta = delta;
}

nakarlsson, 2020-03-03:
@farawayfromhome, can we close this issue?

farawayfromhome, 2020-03-03:
Okay.Sorry to delay.

closed 2020-03-03 by nakarlsson