OpenEtherCATsociety/SOEM #320: ecx_SDOread checks response for index, but not subindex

hoolie, 2019-07-16:
As stated in the title, the ecx_SDOread function verifies that the response contains the correct index, but does not check the subindex:

https://github.com/OpenEtherCATsociety/SOEM/blob/abbf0d42e38d6cfbaa4c1e9e8e07ace651c386fd/soem/ethercatcoe.c#L175
  
<br/><br/>
The ecx_SDOwrite method checks the subindex :

https://github.com/OpenEtherCATsociety/SOEM/blob/abbf0d42e38d6cfbaa4c1e9e8e07ace651c386fd/soem/ethercatcoe.c#L378 and
https://github.com/OpenEtherCATsociety/SOEM/blob/abbf0d42e38d6cfbaa4c1e9e8e07ace651c386fd/soem/ethercatcoe.c#L448
<br/>
**Is there a reason why the subindex is not checked in the read function?**

ArthurKetels, 2019-07-20:
This is a left-over fix for some very old slaves that forgot to set the sub-index on the SDOread. I guess that with the current slave conformance test being mandatory these slaves have died out by now. 

But for the same reason the test is not that important. The probability finding a slave that outputs the wrong object in response to an upload request is near to zero. And mix-ups with simultaneous requests are prevented by the frame numbering.

hoolie, 2019-08-07:
Thank you for the clarification.

closed 2019-08-07 by hoolie