OpenEtherCATsociety/SOEM #321: Combining RX & TX frame fails without a packet filter

bjornlindAXIS, 2019-07-18:
OS: Windows 10
Project: Simple_test & Slaveinfo

When I try to auto-config the slaves with _ec_config()_ I see in Wireshark that the returning frame has detected the correct number of slaves (wkc = 3 in my specific case). However as _ec_config()_ finishes, the following is the output:
```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{87F5F304-9223-413F-AE17-E9A8B346621E} succeeded.
Working counter = 0
No slaves found!
End slaveinfo, close socket
End program  
```
So it seems that the transmitted and read frame are not combined properly in the _ecx_inframe()_ function in **nicdrv.c**.

My solution to this is currently to add a packet capture filter in **nicdrv.c** to ignore any packets coming from 01:01:01:01:01:01. This is undesirable though since I don't want to alter the source code. The following code is inserted after the psocket has been opened:
```
   struct bpf_program bpf_program;
   char filterString[] = "ether src not 01:01:01:01:01:01";
   pcap_compile(*psock, &bpf_program, filterString, 0, 0);
   if (pcap_setfilter(*psock, &bpf_program)) 
   {
	   printf("Could not set up capture filter\n");
   }
```

Here is the output from **Slaveinfo** with the filter:
[Slaveinfo output with packet filter.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3406823/Slaveinfo.output.with.packet.filter.txt)

Here are wireshark captures both with and without the filter:
[wstraces_packetcombineissue.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3406833/wstraces_packetcombineissue.zip)

Appreciate any input into this, thank you!

ArthurKetels, 2019-07-20:
In nicdrv.c we open winpcap with:

<code>
   *psock = pcap_open(ifname, 65536, PCAP_OPENFLAG_PROMISCUOUS |
                                     PCAP_OPENFLAG_MAX_RESPONSIVENESS |
                                     PCAP_OPENFLAG_NOCAPTURE_LOCAL, -1, NULL , errbuf);
</code>

The PCAP_OPENFLAG_NOCAPTURE_LOCAL should take care of filtering the outgoing packets. I know that Win10Pcap does not support that flag. So either use the original WinPcap or NPcap.

From similar errors reported it could be you are opening some virtual or intermediate driver. This will prevent correct capturing also. 

I you can't find the source of the error, the proposed change is of course a solution. 

Many users have no problems with windows10 and SOEM.

bjornlind92, 2019-07-22:
Switching to NPcap works, thanks for your swift reply!

closed 2019-07-22 by bjornlind92