OpenEtherCATsociety/SOEM #322: Getting error while integrating with ros node in C++.

MRIAC, 2019-07-26:
Hi all,

I am trying to integrate soem library to ros node in C++ code. so, i am getting undefined reference error while catkin build.
please suggest some solution, i have been struggling from last 1 week.

Thanks

nakarlsson, 2019-07-26:
Have you tried the SOEM ros package? http://wiki.ros.org/soem
Not sure what is needed, if there is a simple and generic adoption to enable/fix your C++ linking issue,  please provide a PullRequest.

MRIAC, 2019-07-29:
thanks @nakarlsson  i had tried with SOEM ros package, But still i am facing same
 issue

MRIAC, 2019-07-29:
i am getting fatal **error: hpeif2.h: No such file or directory** while putting catkin build command. Do you have any idea about this or else can you suggest me the link to get the hpeif2.h header file..???

nakarlsson, 2019-07-29:
Do you compile for InTime? If so, the InTime port is for version 4 and the IF using the header you can’t find. If that don’t work you might need to create a new/updated InTime port.

If you don’t compile for InTime you need to study the documentation provided before writing issues. I know for a fact that SOEM works with ROS on Linux since we’ve customers using it.

MRIAC, 2019-07-29:
@nakarlsson i have tried with SOEM used with ROS on linux but getting  u**ndefined reference to `osal_thread_create'
collect2: error: ld returned 1 exit status** error.

nakarlsson, 2019-07-29:
Have you built the SOEM lib for Linux according to the readme?
Have you added the SOEM lib to the list of libraries to link?

MRIAC, 2019-08-02:
@ sorry for late reply...please find below error which i am facing.

Profile: default
Extending: [env] /opt/ros/kinetic
Workspace: /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws
Build Space: [exists] /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/build
Devel Space: [exists] /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/devel
Install Space: [unused] /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/install
Log Space: [missing] /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/logs
Source Space: [exists] /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src
DESTDIR: [unused] None
Devel Space Layout: linked
Install Space Layout: None
Additional CMake Args: None
Additional Make Args: None
Additional catkin Make Args: None
Internal Make Job Server: True
Cache Job Environments: False
Whitelisted Packages: None
Blacklisted Packages: None

Workspace configuration appears valid.
NOTE: Forcing CMake to run for each package.

[build] Found '1' packages in 0.0 seconds.
[build] Updating package table.
Starting >>> catkin_tools_prebuild
Finished <<< catkin_tools_prebuild [ 1.5 seconds ]
Starting >>> ethercat_soem

Errors << ethercat_soem:make /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/logs/ethercat_soem/build.make.000.log
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp: In member function ‘void Epos_interaction::initialized_thread()’:
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp:595:73: warning: converting from ‘void (Epos_interaction::)()’ to ‘void’ [-Wpmf-conversions]
osal_thread_create(&thread1, 128000, (void*)&Epos_interaction::ecatcheck, (void*) &ctime);
^
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp: In member function ‘void Epos_interaction::initialized_thread()’:
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp:595:73: warning: converting from ‘void (Epos_interaction::)()’ to ‘void’ [-Wpmf-conversions]
osal_thread_create(&thread1, 128000, (void*)&Epos_interaction::ecatcheck, (void*) &ctime);
^
CMakeFiles/ethercat_soem.dir/src/final.cpp.o: In function Epos_interaction::initialized_thread()': final.cpp:(.text+0x5c13): undefined reference to osal_thread_create'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/devel/.private/ethercat_soem/lib/ethercat_soem/ethercat_soem] Error 1
make[1]: *** [CMakeFiles/ethercat_soem.dir/all] Error 2
make: *** [all] Error 2
cd /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/build/ethercat_soem; catkin build --get-env ethercat_soem | catkin env -si /usr/bin/make --jobserver-fds=6,7 -j; cd -
..............................................................................................................................................
Failed << ethercat_soem:make [ Exited with code 2 ]
Failed <<< ethercat_soem [ 4.6 seconds ]
[build] Summary: 1 of 2 packages succeeded.
[build] Ignored: None.
[build] Warnings: None.
[build] Abandoned: None.
[build] Failed: 1 packages failed.
[build] Runtime: 6.2 seconds total.
[build] Note: Workspace packages have changed, please re-source setup files to use them.
atugv@atugv-Precision-M4800:~/Manpack_Drive_code/ethercat_motor/catkin_ws

mgruhler, 2019-08-02:
@nakarlsson I've taken over the maintainment of the SOEM ROS wrapper package from @smits Feel free to redirect any questions about ROS to the mgruhler/soem Repo.

@MRI

closed 2019-08-02 by MRIAC

nakarlsson, 2019-08-02:
Sounds good @mgruhler