OpenEtherCATsociety/SOEM #325: integration of soem c libarary to ROS C++ node.

MRIAC, 2019-08-01:
since i am not getting any idea how to integrate soem library which is written in C, i have written my own C++ ROS node to publish the topic and i wanted to link soem library with my C++ node.

i have already tried with soem library for ROS but i am getting undefined reference error...so anybody can suggest the entire procedure for the same.

MRIAC, 2019-08-01:
--------------------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws
--------------------------------------------------------------------------------------------
Build Space:        [exists] /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/build
Devel Space:        [exists] /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/devel
Install Space:      [unused] /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/install
Log Space:         [missing] /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/logs
Source Space:       [exists] /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                                                                                    
[build] Updating package table.                                                                                                               
Starting  >>> catkin_tools_prebuild                                                                                                           
Finished  <<< catkin_tools_prebuild                [ 1.5 seconds ]                                                                            
Starting  >>> ethercat_soem                                                                                                                   
______________________________________________________________________________________________________________________________________________
Errors     << ethercat_soem:make /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/logs/ethercat_soem/build.make.000.log                
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp: In member function ‘void Epos_interaction::initialized_thread()’:
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp:595:73: warning: converting from ‘void (Epos_interaction::*)()’ to ‘void*’ [-Wpmf-conversions]
          osal_thread_create(&thread1, 128000, (void*)&Epos_interaction::ecatcheck, (void*) &ctime);
                                                                         ^
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp: In member function ‘void Epos_interaction::initialized_thread()’:
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp:595:73: warning: converting from ‘void (Epos_interaction::*)()’ to ‘void*’ [-Wpmf-conversions]
          osal_thread_create(&thread1, 128000, (void*)&Epos_interaction::ecatcheck, (void*) &ctime);
                                                                         ^
CMakeFiles/ethercat_soem.dir/src/final.cpp.o: In function `Epos_interaction::initialized_thread()':
final.cpp:(.text+0x5c13): undefined reference to `osal_thread_create'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/devel/.private/ethercat_soem/lib/ethercat_soem/ethercat_soem] Error 1
make[1]: *** [CMakeFiles/ethercat_soem.dir/all] Error 2
make: *** [all] Error 2
cd /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/build/ethercat_soem; catkin build --get-env ethercat_soem | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..............................................................................................................................................
Failed     << ethercat_soem:make                   [ Exited with code 2 ]                                                                     
Failed    <<< ethercat_soem                        [ 4.6 seconds ]                                                                            
[build] Summary: 1 of 2 packages succeeded.                                                                                                   
[build]   Ignored:   None.                                                                                                                    
[build]   Warnings:  None.                                                                                                                    
[build]   Abandoned: None.                                                                                                                    
[build]   Failed:    1 packages failed.                                                                                                       
[build] Runtime: 6.2 seconds total.                                                                                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.
atugv@atugv-Precision-M4800:~/Manpack_Drive_code/ethercat_motor/catkin_ws

mgruhler, 2019-08-02:
@nakarlsson @MRIAC this is now really duplicating #322 
can be closed

nakarlsson, 2019-08-02:
As agreed with ROS SOEM, handle at https://github.com/mgruhler/soem

closed 2019-08-02 by nakarlsson