OpenEtherCATsociety/SOEM #326: I think I broke settings the working slave device.

rasimckc, 2019-08-02:
He all,
I was connected to a 'Selema drive' before and did a 'simple test'.I didn't have a problem.I didn't run into a problem, but I couldn't test later when I connected.I may have written the wrong address when I first connected. So I think I've corrupted some settings.I reset the parameters from the drive, but it hasn't improved.

When I first connected to the drive with ethercat explorer, the name I saw was 'Selema_Device-1001'.In this case I could communicate and there was no problem.
[image]
But now when I connect, the result is a complex name.
[image]
The photo was taken from simple ethercat explorer.When I searched with 'Soem' I got the same result.
This is the name of the network when I searched.
[image]
I can't test.
[image]
How do you think that happened?How do I fix it?
Thank you from now.

nakarlsson, 2019-08-02:
Have you written the EEPROM? Eg filled it with a bad EEPROM data?

rasimckc, 2019-08-02:
yes i wrote.I think I  filled eeprom with bad data.

nakarlsson, 2019-08-02:
Then you need to reprogram it with correct data. Either with the actual EEPROM image via eepromtool or similar Or TwinCAT with help of the ESI.

rasimckc, 2019-08-02:
@nakarlsson I fixed it through twincat as you said.Thank you

rasimckc, 2019-08-02:
@yusufselimkaratas   Would you send Turkish mail to? :D cekic36@gmail.com

closed 2019-08-02 by rasimckc