OpenEtherCATsociety/SOEM #327: ALL_BUILD Unable to start program

yusufselimkaratas, 2019-08-02:
Hello everyone,

I took used cmake GUI and got output files and opened directly with VS 2017. Without any change I tried to use Local Windows Debugger and I got an error like  "Unable to start program ' C.\users\yusufSelim\Desktop\OUT\Debug\ALL_Build'" . It says that System can not find the file. Also I observed that Debug does not have any All Build. How can I solve this?

Thanks.

rasimckc, 2019-08-02:
He,
right click then set as startUp project
[image]
Will you try that?

yusufselimkaratas, 2019-08-02:
Amazing. It works. If you come to Turkey, just write me. I own you a lunch. Thanks.

closed 2019-08-02 by yusufselimkaratas

rasimckc, 2019-08-02:
rica ederim Yusuf.