OpenEtherCATsociety/SOEM #328: Simple_test argc error

yusufselimkaratas, 2019-08-02:
Hello everyone,

I run simple_test and got the output given in below. It seems that argc does not become bigger than 1. What may be the reason behind that?

Thanks.

SOEM (Simple Open EtherCAT Master)
Simple test
Yusuf SelimUsage: simple_test ifname1
ifname = eth0 for example
End program

C:\Users\Yusuf Selim\Desktop\OUT\test\linux\simple_test\Debug\simple_test.exe (process 13792) exited with code 0.
Press any key to close this window . . .

yusufselimkaratas, 2019-08-02:
[image]

Actually sometyhing is going on but what?

yusufselimkaratas, 2019-08-02:
In my code argc and argv[] does not change ? So it does not enter in if condition. How can I solve this? What may be the reason behind this?

yusufselimkaratas, 2019-08-02:
[image]

yusufselimkaratas, 2019-08-02:
[image]

nakarlsson, 2019-08-02:
You need to provide args to the program, either via the command line or the debug settings from VisualStudio.

yusufselimkaratas, 2019-08-02:
Hello, Thanks very much but It did not work

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
### interface eth1 could not open with pcap
No socket connection on eth1
Excecute as root
End program

C:\Users\Yusuf Selim\Desktop\OUT\test\linux\simple_test\Debug\simple_test.exe (process 9396) exited with code 0.
Press any key to close this window . . .

I have tried all eth0,eth1 etc. Could not solve the problem. 

int main(int argc, char *argv[])
{
   printf("SOEM (Simple Open EtherCAT Master)\nSimple test\n");   
   if (argc >1)
   {
      /* create thread to handle slave error handling in OP */
      //pthread_create( &thread1, NULL, (void *) &ecatcheck, (void*) &ctime);
      osal_thread_create(&thread1, 128000, &ecatcheck, (void*) &ctime);
      /* start cyclic part */
      simpletest(argv[1]);
   }
   else
   {
      printf("Usage: simple_test ifname1\nifname = eth0 for example\n");
   }
   printf("End program\n");
   return (0);
}

here is the main function. I dont know. Where is the error?

yusufselimkaratas, 2019-08-02:
As far as I understand I can not set the eth name correctly. Is there anywhere I can learn my network name?

nakarlsson, 2019-08-02:
Yoy should use the name(s) from the screen you attached. \Device\NPF... something. It is the key WinPCAP use to open an interface

yusufselimkaratas, 2019-08-05:
I remember I have tried that but It did not worked. I will try it again today.

yusufselimkaratas, 2019-08-05:
@nakarlsson Thanks. I tried it again and It has worked.

closed 2019-08-05 by yusufselimkaratas