OpenEtherCATsociety/SOEM #329: ec_SDOread returns 0xCCCC

yusufselimkaratas, 2019-08-06:
Hello everyone, 

Is there any explaination of reading 0xCCCC everytime regarless of read data with this code? as far as I understood, I can not read the data and assign to read_data . So program prints 0xCCCC which is default parameter.

Thanks.

for (int slavecounter = 1; slavecounter <= ec_slavecount; slavecounter++)

	{
		int retval2;
		uint16 read_value;
		retval2 = 0;
		uint16 read_index;
		uint8 read_subindex;
		int read_length;

		read_index = 0x2143;
		read_subindex = 0x00;

		retval2 += ec_SDOread(slavecounter, read_index, read_subindex, FALSE, &read_length, &read_value, EC_TIMEOUTRXM);
		printf("%X has been read from Index:%X, Subindex:%X --length:%X \n", read_value, read_index, read_subindex,read_length);
	}

yusufselimkaratas, 2019-08-06:
retval returns 0

nakarlsson, 2019-08-06:
if you read the documentation you would know read_length is an in/out parameter and that you should pass the parameter length in, eg sizeof read_value if the index is uint16?
read_length return actual length read.

yusufselimkaratas, 2019-08-07:
Thanks @nakarlsson , 

I have converted my code tobelow and it has worked.

`read_index = 0x220A;
		read_subindex = 0x00;
		
		uint16 read_value;
		int read_length;

		read_length = sizeof(read_value); read_value = 0;

		retval2 += ec_SDOread(slavecounter, read_index, read_subindex, FALSE, &read_length, &read_value, EC_TIMEOUTRXM);
		printf("RetVal : %d\n", retval2);
		printf("%X has been read from Index:%X, Subindex:%X --length:%X \n", read_value, read_index, read_subindex,read_length);`

closed 2019-08-07 by yusufselimkaratas