OpenEtherCATsociety/SOEM #333: Can't reach operational state - EEPROM no access

cianpdx, 2019-08-09:
Hi team - is this show stopping issue? 

still wrapping head around SOEM but wanted to check this before diving in. 

I have increased timeouts 2x from reading other issue logs - with no change. 

```
Link Check...OK
ec_init on -map succeeded.
wkc = 0
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=11 StatusCode=  50 : EEPROM no access

Slave:1
 Name:VDM MFC
 Output size: 40bits
 Input size: 72bits
 State: 17
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000602 ID: 0000012c Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   5 F:00010064 Type:3
 SM3 A:1180 L:   9 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   5 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000005 Ll:   9 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 2f FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```

nakarlsson, 2019-08-10:
I’m not familiar with that error, you need to consult the slave manual on how to fix/handle it.
Sounds like the PDI or ECAT don’t get EEPROM access when it wants it. SOEM got functions to handle EEPROM access, ec_eeprom2pdi and ec_eeprom2master.

cianpdx, 2019-08-12:
Thanks - does no-op state prevent reading from 1A01 PDO - or is that device specific too?

nakarlsson, 2019-08-13:
Your slave report state INIT + E , in INIT you can pretty much only access the ESC registers and via the ESC the EEPROM.
So, before fixing the the error you can’t do much. You should find out more about that error, no shortcuts available.

closed 2019-08-13 by cianpdx