OpenEtherCATsociety/SOEM #334: Is it possible to broadcast the same data to a slave group without duplicating data for each slave in that group

NPScott, 2019-08-16:
I have been playing around with this SOEM library and managed to quite easily set up a network of slaves, reading and writing data via IOmap.  Each slave has a separate area of memory in the IOmap and every ethernet datagram contains all this data.  In my particular case I want to send the same data to a group of slaves; is there a way to do this without duplicating the same data in the IOmap / datagram, as this seems really inefficient?

Thanks,
Nathan

nakarlsson, 2019-08-16:
Not out of the box, but you can explicitly program fmmus/syncmanagers to listen to the same logic address. You can program the fmmus with fixed position address write/FPWR

NPScott, 2019-08-16:
Thank you for such a quick response.
I presume this programming needs to be done at the slave device or can the master tell the slave to adjust this?  I've looked around the slave source code but there seems no obvious way this is done.

nakarlsson, 2019-08-16:
The master set the values of the slave ESC Hw to do this

NPScott, 2019-08-16:
Ok - I guess this can only be done at start up or can this be changed at any time?  Ideally we would like to change this in response to some event at the slave.

Once again thank you for your help.

nakarlsson, 2019-08-20:
If this data is not part of normal process data, eg not used or configured during startup or normal run-time. I guess it is up to you, when, how and if you set it up. Don't think this type of operation is covered by the EtherCAT standard.

NPScott, 2019-08-23:
OK - may be the best way to proceed would be to send some common data to all slaves in the group mixed with data which is slave# specific.  Is this possible with SOEM, as I did not see an API for setting this up?  Looking at the EtherCAT standard this would be considered normal operation...?  So at start the master could specify say the first 20 bytes in the IOmap is common to all slaves and then 40 bytes boundaries for each of the connected slaves.  This way we do not need to resend the first 20 bytes to each slave via the IOmap mechanism.

Sorry lots of questions there - I am just trying to see if this route is worth considering for our project.

nakarlsson, 2019-08-24:
By default SOEM create an IOmap from PDOs avaliable in the slaves SII/CoE PDO mappings, no API is available to manipulate that to do what you want.
However, all local data like the IOmap and slave configurations are available and there are lots of functions available to read/write configurations of/to the slaves. These combined can do pretty much what you want within the limits of the EtherCAT HW.
I suggest you dig deep in how EtherCAT and SOEM works and you should be able to create a solution.

But, what are you realy trying to achive. What is the actual problem your solving? EtherCAT is quite fast and efficient out of the box.
Second,  study the concept of slave to slave communication to better understand how you can share data.

NPScott, 2019-08-24:
We broadcast audio and visual data to receivers (slaves) in a closed LAN network.  The user can select a channel and they are then presented with audio and video data based on their language selection.  Now if for example 10 users select Swedish then they will all receive the same broadcast at the same time, whist the other users receive this data in their language selection.  We have been doing this for a few years via a Multicast Ethernet solution and this does work well.  However, the motivation for looking at EtherCAT is cost reduction as we would not need a full Ethernet switch at every device.

From my work so far I can see SOEM out of the box lets you easily send and receive large amounts of data via the IOmap but an entry for each slave.  It is not uncommon to have 100+ slaves in the LAN, so currently this would require a huge IOmap.  However, there is a lot of common data so if the mapping could change based on user language selection then we would be in business.  Does this make sense or am I just looking at it in the wrong way?

nakarlsson, 2019-08-24:
You have to consider that EtherCAT is a standard fieldbus that has rules that are there to ensure interoperability between devices from many different vendors as every other fieldbus, you have a state machine , certain way of designing things and mostly you control I/O and drives.

However I think there are a possablity you can make it work, but asking question here tend to give you answers based on the standard and based on common industrial networks/Automation solutions. SOEM out of the box is based on the later type of EtherCAT networks that typically don’t work as yours.

I think:
In stead of designing the EtherCAT network per I/O of slaves, you should have a network based on languages, and make the slaves listen to the correct language. This would need re-programing the EtherCAT HW on user selection, and maybe make use if additional SyncManagers and FMMUs, beside the normal 4+2. As I Said in previous post, you need to dig deep.

ArthurKetels, 2019-08-26:
I have done something similar a decade ago. Using EtherCAT as a large PA system. Multiple sources and destinations needed to be cross connected. I would suggest to use extra mailboxes as data transport and use the 3-buffer IO map to convey state across the network. So each slave would need 2 extra SM and 2 extra FMMU to work. Then you need to define your own VoE protocol over the mailboxes. The big benefit is that this will stay within the EtherCAT specification and is compatible with "standard" slaves on the network.

If you want more details perhaps it is better to this via PM. It would be way off topic here.

NPScott, 2019-08-26:
Hi Arthur,
Thank you for your reply and yes happy to take this off the forum, how do I contact you?
Thanks,
Nathan

ArthurKetels, 2019-08-27:
soem(at)smfk(dot)nl

ArthurKetels, 2019-08-29:
I got no response?

nakarlsson, 2020-03-09:
Can we close this?

nakarlsson, 2020-05-09:
Inactive

closed 2020-05-09 by nakarlsson