OpenEtherCATsociety/SOEM #336: ecatcheck( void *ptr )

lyhawk2007, 2019-08-30:
If the connection is more slave equipments, one is off-line, the calling function ecatcheck (void * PTR), make the equipment comes back on-line. but sometimes can be back on-line, sometimes can't it, print information "ERROR: slave 1 is in SAFE_OP + ERROR, attempting an ack". This kind of situation, equipment in off-line forever, if can make equipment on-line again?Why and how to avoid it.My ESI is configured as follows: 
"<PreopTimeout>2000</PreopTimeout>
<SafeopOpTimeout>65535</SafeopOpTimeout>
<BackToInitTimeout>5000</BackToInitTimeout><BackToSafeopTimeout>200</BackToSafeopTimeout>"

lyhawk2007, 2019-08-30:
PreopTimeout  2000  PreopTimeout
							SafeopOpTimeout 65535 SafeopOpTimeout
							BackToInitTimeout 5000 BackToInitTimeout
							BackToSafeopTimeout 200 BackToSafeopTimeout

closed 2019-08-30 by lyhawk2007

ArthurKetels, 2019-08-30:
Sorry, not enough information to give a sensible answer. 

If for example you have a slave that needs synchronous operation and your computer has timing issues an ack from SOEM to the slave will not help and the slave will go to error state again. You have to read the error state of the slave to find out the cause.

So to give a better answer I need the slave types, a wireshark trace and the error status.

ecatcheck() is intended to solve loss of communication or slave resets, not for other issues.