OpenEtherCATsociety/SOEM #339: Is it possible to run SOEM on Cortex-M device with Ethernet support?

AntonioMichaelAlvarez, 2019-09-18:
Hi! I have a development board with STM32F746 MCU and onboard Ethernet. Is it possible to run SOEM on it?
Are there any Cortex-M devices capable of running SOEM?
rt-labs FAQ [page](https://rt-labs.com/ethercat/faq/) says it is possible to implement master on STM32F4. If so, are there any reference projects to study?

nakarlsson, 2019-09-18:
Not for STEM32F4 here, but similar for imx6 with MAC named FEC.
https://github.com/OpenEtherCATsociety/SOEM/blob/master/oshw/rtk/fec/fec_ecat.c , is running on a imx6 FEC and our RTOS rt-kernel.

AntonioMichaelAlvarez, 2019-09-18:
@nakarlsson 
>  similar for imx6

i.MX6 is Cortex-A9, but STM32 MCUs are based on Cortex-M architecture (Cortex-M4F for STM32F4 and Cortex-M7F for STM32F7). 

I was doing research if STM32F7 is capable of running EtherCAT Master and stumbled upon rt-labs FAQ featuring possible implementation on STM32F4 device. SOEM documentation doesn't directly state support for STM32 MCUs. Should I contact rt-labs on this subject?

nakarlsson, 2019-09-18:
I work at rt-labs so you're already in contact.
Yes, but the idea is the same, regardless of A9, M4F or M7F. As a ref. we have doone projects on Infineons XMC48 which is a Cortex M4F. Of course, you need to tweak SOEM abit to be able to fit in 300-400k RAM, but doable.

closed 2019-10-03 by AntonioMichaelAlvarez