OpenEtherCATsociety/SOEM #341: SOEM to ease CI tests

calvama, 2019-10-09:
Would it be possible to have a record&playback feature in SOEM to ease regression testing on the cloud without the need of the hardware?

Feature request initially started here:
https://github.com/mgruhler/soem/issues/31

ArthurKetels, 2019-10-10:
I have read the rationale for the request in the link you mentioned. And while I do understand the advantages for such functionality I have a hard time thinking of an implementation that would actually work. 

EtherCAT networking is by nature very dynamic. Both on the master<->slave and application<->master side. The common use case if for some kind of control feedback loop. The slightest change in one of the implementations or just plain randomness will lead to very different data packets being generated.

I could imagine a tool that uses a wireshark capture as a play back source. Packets from the master and packets returning from the slaves are easy to distinguish by their fake MAC addresses. So the tool waits for a master packet to arrive, compares it with the list and returns the corresponding slave answer. This can be done completely independent from SOEM, and is relatively easy to implement. But the actual usefulness would be rather limited in my opinion.

calvama, 2019-10-15:
Using wireshark was our first idea, but it requires modifying the packets (since there are timestamps, and probably some more stuff, sorry I am not an expert on the protocol). So, at least SOEM already has structures to handle packages.
One possibility is to create this tool based on SOEM, and separated from it, but then there should be two builts, one for testing and one for release (not too elegant).
Considering modifying SOEM, instead of playing back at package management level, it could be done at inteface level, so SOEM returns each function call as if the actual slave had answered.

We can spend some time doing this if there is an agreement on its usefulness.

nakarlsson, 2020-03-09:
ref issue is closed, so we'll close this one as well

closed 2020-03-09 by nakarlsson