OpenEtherCATsociety/SOEM #344: LAN9252 SlaveInfo not found

Davidinos, 2019-10-14:
Goodmorning to all,
I write you since I'm trying to use the SOEM library on my Beaglebone black board. As slave device I use an Arduino Uno with an ethercat shield produced by https://www.bausano.net/it/hardware/arduino-ethercat/arduino-easycat.html which use the LAN9252 Microchip's IC.

As kernel I used the one with xenomai installed (available here: https://github.com/RobertCNelson/ti-linux-kernel-dev/tree/ti-linux-xenomai-4.14.y), I set a static IP on eth0 as 192.168.2.130 and installed the SOEM library as explained in README file.
The answer to command ifconfig is:
eth0: flags=-28349<UP,BROADCAST,RUNNING,PROMISC,MULTICAST,DYNAMIC>  mtu 1500
        inet 192.168.2.130  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::eaeb:11ff:fe27:b92b  prefixlen 64  scopeid 0x20<link>
        ether e8:eb:11:27:b9:2b  txqueuelen 1000  (Ethernet)
        RX packets 443  bytes 64423 (62.9 KiB)
        RX errors 0  dropped 5  overruns 0  frame 0
        TX packets 1043  bytes 367526 (358.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 45
When I run the salveInfo application, I get the following answer:
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
[ 1695.543215] device eth0 entered promiscuous mode
ec_init on eth0 succeeded.
No slaves found!
End slaveinfo, close socket
End program

For completeness this is the answer of Arduino sketch when I power it on:
EasyCAT - Generic EtherCAT slave
CUSTOM MODE
1 Byte Out
8 Byte In
Sync = ASYNC
Detected chip 9252  Rev 1
initialized

I did a wireshark capture connecting the LAN9252's ethernet port OUT to my PC and I get the
following:
[Capture.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3724249/Capture.zip)

Have you got any suggestion of the test to do?
Thank you

ArthurKetels, 2019-10-14:
By connecting the LAN9252 output to your PC you have broken the EtherCAT ring. All packets send are no longer received back by the master. Disconnect the out port and capture wireshark on your beaglebone directly.

Davidinos, 2019-10-15:
Hello ArturKetels,
thank you for your answer.

For a better understanding my connections are similar to the ones shown in the following video: https://www.youtube.com/watch?v=jGTZ8ueQfGg. 
The only difference is that I'm using a beaglebone black not a raspberry pi 3. There isn't a closed ring, but it has to work. I tried to connect beaglebone to the LAN9252 both with a standard ethernet cable and a cross-talk one but I get the same result.
I tried to connect beaglebone (static ip 192.168.2.130) directly to PC (static ip 192.168.2.10), launch slaveInfo application and I get the following Wireshark capture:
[Capture2.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3729485/Capture2.zip).

Thank you for your help.

Davidinos, 2019-10-16:
Hello ArturKetels,
thank you for your answer.

For a better understanding my connections are similar to the ones shown in the following video: https://www.youtube.com/watch?v=jGTZ8ueQfGg.
The only difference is that I'm using a beaglebone black not a raspberry pi 3. There isn't a closed ring, but it has to work. 
I've remove the PC connection to LAN9252 OUT port, and for monitoring ethernet traffic, I installed on beaglebone tcpdump.
I got the following result:

15:31:21.536039 Out 01:01:01:01:01:01 (oui Unknown) ethertype Unknown (0x88a4), length 31:
        0x0000:  0d10 0801 0000 0301 0100 0000 0000 00    ...............
15:31:21.543456 Out 01:01:01:01:01:01 (oui Unknown) ethertype Unknown (0x88a4), length 31:
        0x0000:  0d10 0802 0000 2001 0100 0000 1100 00    ...............
15:31:21.551416 Out 01:01:01:01:01:01 (oui Unknown) ethertype Unknown (0x88a4), length 31:
        0x0000:  0d10 0803 0000 2001 0100 0000 1100 00    ...............
15:31:21.559399 Out 01:01:01:01:01:01 (oui Unknown) ethertype Unknown (0x88a4), length 32:
        0x0000:  0e10 0704 0000 0000 0200 0000 0000 0000  ................
15:31:21.568786 Out 01:01:01:01:01:01 (oui Unknown) ethertype Unknown (0x88a4), length 32:
        0x0000:  0e10 0704 0000 0000 0200 0000 0000 0000  ................
15:31:21.575444 Out 01:01:01:01:01:01 (oui Unknown) ethertype Unknown (0x88a4), length 32:
        0x0000:  0e10 0704 0000 0000 0200 0000 0000 0000  ................

Thank you for your help.

ArthurKetels, 2019-10-16:
It is a bit hard to read the tcpdump output, but it looks like there are no packets returned to the beaglebone. Either there is a filtering rule active that kills all incoming EhterCAT packets or the slave hardware is not functioning properly.

Things you can test:
a) replace beaglebone by a PC/laptop running SOEM
b) replace slave by other known-to-work slave
c) connect beaglebone<-->LAN9252(OUT) instead of (IN)

Davidinos, 2019-10-16:
Hello ArthurKetels,
thank for your suggestion. I tried to connect PC (with SOEM) to LAN9252 and it works, so we can state that the slave is working properly. Below the answer:

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth0 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:myCustomDavideSlave
 Output size: 96bits
 Input size: 88bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:0.1.0.0
 Configured address: 1001
 Man: 0000079a ID: deadbeef Rev: 00000001
 SM0 A:1000 L:  12 F:00010064 Type:3
 SM1 A:1200 L:  11 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  12 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000c Ll:  11 Lsb:0 Leb:7 Ps:1200 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

Concerning the beaglebone, even trying to connect it to OUT port the result doesn't change.
Have you got any advices?

Thank you very much.

ArthurKetels, 2019-10-16:
Look at the linux kernel version you use. There has been a bug in the network stack last year that prevented outputting the correct frames. It can also be there are some firewall rules active that prevent frames returning to SOEM.

Davidinos, 2019-10-16:
Hello ArthurKetels,
I wanted to post these screenshots of kernel menuconfig before your answer, but I did it too late.
Could you check if there is some module that need to be mounted or that they are missing in networking option menu?

[image]
[image]

Regarding the the kernel version is the following:
Linux arm 4.14.108-ti-xenomai-r120 #1 SMP PREEMPT Thu Oct 10 09:31:35 CEST 2019 armv7l GNU/Linux

Concerning the firewall,  it looks like that there is no firewalld, nor ufw, installed. The iptables status is the following:

debian@arm:/sbin$ sudo /sbin/iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination 

In the meantime I'll try to ask to Robert Nelson's repository if he's got some suggestions.
Thank you.

Davidinos, 2019-10-18:
Hello ArthurKetels,
Robert Nelson answered me and he asked me if you have a reference commit for the issue concering the network stack last year that prevented outputting the correct frames.
Do you know it?
Thank you in advanced.

Davidinos, 2019-10-21:
Hello ArthurKetels,
I solved the problem, basically I needed to downgrade the kernel version to:
Linux arm 4.4.113-ti-xenomai-r149.1 #1 SMP Mon Oct 21 11:55:55 CEST 2019 armv7lx,
don't know why with version 4.14 I had problem.
For the moment I close the issue, but I'LL BE BACK.

Best Regards and thank you.

closed 2019-10-21 by Davidinos

Bipin903, 2024-04-05:
Hi @Davidinos can you guide me how to send and receive data from arduino using SOEM .

Davidinos, 2024-04-05:
> Hi @Davidinos can you guide me how to send and receive data from arduino using SOEM .

Hi, I can suggest you this one. 
https://www.bausano.net/it/hardware/arduino-easycat.html
with this explanation in english
https://www.bausano.net/images/arduino-easycat/Easy_Master_UserManual.pdf
It can be a starting point.

However I didn't implemented the ethercat master nor with arduino neither with Beaglebone black as explained in the thread.
At the end I adapted the SOEM for an NXP microprocessor.