OpenEtherCATsociety/SOEM #349: ec_config_map seems to overwrite default values on ELMO GOLD CELLO

frriccar, 2019-10-22:
Hello,

I'm starting to use SOEM to drive an ELMO GOLD CELLO driver, and considering it's the first time I'm using both, I'm going slow and trying to understand how they work.
In principle what I'm trying to do is to command the driver in velocity (Profile Velocity Mode) with only SDO communication, but I have noticed a very strange (or maybe not) thing:

- If I try to do it while the slave is in PRE_OP and without configuring the IOmap (ec_config_map), it works and I can see all the default values (acceleration, max speed, etc.) with readSDO;

- As soon as I call the function to initialize the IOmap, all the default values are overwritten (for example acceleration = 1) and it doesn't work anymore.

Here you can see the output:

`SOEM (Simple Open EtherCAT Master)
Slaveinfo
ec_init on eno1 succeeded.
1 slaves found and configured.
Slave 1 has CA? true
PREOP.
Slave: 1 - Read at 0x6083:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Profile acceleration]
Slave: 1 - Read at 0x6084:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Profile deceleration]
Slave: 1 - Read at 0x6085:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Quick stop deceleration]
 ___________________________________ 
IOmap Configured
___________________________________ 
PREOP.
Slave: 1 - Read at 0x6083:0 => wkc: 1; data: 0x0001 (1)	[Profile acceleration]
Slave: 1 - Read at 0x6084:0 => wkc: 1; data: 0x0001 (1)	[Profile deceleration]
Slave: 1 - Read at 0x6085:0 => wkc: 1; data: 0x0001 (1)	[Quick stop deceleration]

`
Is it a standard behaviour?
I think it's quite strange, because this mean that for every application, once you have configured the IOmap, you have also to set again all the parameters of a specific Operation Mode (?)
I have already read about this strange behavior in the issue  #245 but it doesn't explain which is the real problem on this values changing.

Thank you.

ArthurKetels, 2019-10-24:
I suggest you read a bit more on how EtherCAT actually works. The process-data exchange is the core of EtherCAT. In OP all communication runs through that protocol. Because it is the most efficient way of communication. The SDO protocol is only used for setting or reading parameters before going to OP.

Perhaps another good suggestion is to first start with a simple slave that has only some digital IO.

frriccar, 2019-10-28:
Hello @ArthurKetels 
Thank you very much for your reply. I know that in OP all the communication is done via PDO (I have experience with CanOpen), I was just becoming familiar with the SOEM and I was surprised to not be able to do a simple motor movement through SDO.

Anyway now it seems to work fine.
Thank you.

closed 2019-10-28 by frriccar