OpenEtherCATsociety/SOEM #350: How to configure SDO SyncManager

Davidinos, 2019-10-25:
Goodmorning to all,
I'm using a beaglebone as Ethercat master and an ethercat slave which uses a LAN9252 and an Arduino. The slave info with -map reports the following:
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
[ 1663.491009] device eth0 entered promiscuous mode
ec_init on eth0 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:TestEasyCAT_Custom
 Output size: 8bits
 Input size: 64bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:0.1.0.0
 Configured address: 1001
 Man: 0000079a ID: abe00001 Rev: 00000001
 SM0 A:1000 L:   1 F:00010064 Type:3
 SM1 A:1200 L:   8 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000001 Ll:   8 Lsb:0 Leb:7 Ps:1200 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to SII :
  SM0 RXPDO 0x1600 Outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x0005:0x01 0x08 UNSIGNED8    Leds
  SM1 TXPDO 0x1A00 Inputs
     addr b   index: sub bitl data_type    name
  [0x0001.0] 0x0006:0x01 0x10 UNSIGNED16   Analog_0
  [0x0003.0] 0x0006:0x02 0x10 UNSIGNED16   Analog_1
  [0x0005.0] 0x0006:0x03 0x10 UNSIGNED16   Bit16_RisingTestRamp
  [0x0007.0] 0x0006:0x04 0x08 UNSIGNED8    DipSwitches
  [0x0008.0] 0x0006:0x05 0x08 UNSIGNED8    Bit8_FallingTestRamp

As shown, PDO input and output are configured and use SM0 and SM1, but there are no syncManager for SDOs; consequently when I try the method ec_SDOread/ec_SDOwrite, they always fail. Unfortunately, at the moment I cannot modify eeprom content, so I need a way for configuring them. I know that it's possible beacause ethercat allows to configure completely a slave even though it hasn't an eeprom. 
Thank you.

In attachment the ESI xml store in the eeprom:
[LAN9252.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3770635/LAN9252.zip)

mheden, 2019-10-25:
According to your log, your slave does not support CoE.

mheden, 2019-10-25:
Yes, LAN9252 by it self supports mailbox communication (at the lower levels) but the slave also need to implement the higher level (SDO transfer state machine, object dictionary etc.). This is done in the software on the application processor attached to the LAN9252 (the Arduino in your case) . So, if your slave doesn't say it don't support CoE whoever implementing the slave didn't implement it.

General: You can configure the syncmanagers of the slave with whatever values you want by writing to the sync manager registers (0x800-) but since the slave most likely have it's opinion on how the sync managers should be configured it you will most likely don't be able to enter PREOP.

ArthurKetels, 2019-10-25:
SOEM test software also has an eepromtool. With this tool you can read and write eeproms on slaves. But you must supply it with a properly configured binary eeprom file.

Davidinos, 2019-10-25:
Hello ArthurKetels and mheden,
thank you for your answers. Just to recap if I understood correctly. 
1. For ethercat slaves, is mandatory to have an EEPROM. 
2. The minimum eeprom content required is to configure SDOs syncmanagers so that there can be basic communication. For writing the eeprom we can use eepromtools.
3. Once SDOs are active (ec_SDOwrite and ec_SDOread can work), they can then be used to configure PDOs and read whatever parameters of the slave.
4. Object Dictionary is not mandatory, since any value can be accessed with SDOs (ec_readSDO and ec_writeSDO), specifying start address and number of register. Object dictionary simply lets to organize parameters in a structure constituted by index, subindex, etc.
5. there are at least 4 SyncManagers in a slave, where one is used for TxPDO and one is used for RxPDO. Multiple TxPDOs can be assign to a single SM (the same goes for RxPDO). Is there a limit?
6. In my specific case, since there are no SDOs enable, I need to rewrite the memory content for setting them?
Thank you very much for your support.

ArthurKetels, 2019-10-25:
I think you are missing a couple of cogs in your EtherCAT mechanics ;-). SDO means having CoE means having Mailbox means having micro-controller. SM  for mailbox don't do anything without firmware support. That is why we have SOES next to SOEM.

Have a look at : https://www.ethercat.org/download/documents/EtherCAT_Device_Protocol_Poster.pdf

A simple slave with only some GPIO does not need a micro-controller but therefore also does not have mailbox, CoE and SDO support.

Davidinos, 2019-10-28:
Hello ArturKetls,
Thank you for your help and patience for answering my questions. It's not really a piece of cake besides I'm not native english speker. I read and thought deeply the document that you indicated me plus some others:
https://infosys.beckhoff.com/english.php?content=../content/1033/ep4174/3386579467.html&id=
https://github.com/OpenEtherCATsociety/SOES/blob/master/applications/linux_lan9252demo/slave.esx.

This is what I've figured out, please correct me if I'm wrong:
I misbelieved that with SDOs I could access LAN9252 registers (such as INTERRUPT ENABLE REGISTER (INT_EN), DIGITAL I/O OUTPUT DATA REGISTER, etc.), but it's not. 
SDOs are only possible if an OD has been realized and stored in eeprom. It's up to the microcontroller connected to LAN9252 makes available the interest registers or values to ethercat master through OD. Aside the OD, in eeprom there must be stored the configuration of LAN9252 registers.

There are still some things that are misterious to me:
1- If there is no microcontroller and LAN9252 is used as GPIO expander or together with MCP23017, how can ethercat master access data (considering that there are no COE)?
2- In the slave.esx, I couldn't see a specific connection between the OD general entries (like Parameters, software version, Device Name, etc.) and their mailbox SyncManager (on the other hand the link concerning buttons and leds is clear).
3- In slave.esx, address 0x6000, 0x7000 and 0x8000 are referred and they are coherent with Profile-specific objects (0x6000-0xFFFF). But considering that the last LAN9252's addresses are 1000h-1FFFh Process Data RAM, I don't get where they are stored.

Thank you very much for your support.
Regards

mheden, 2019-11-04:
1. In that case the PDOs needs to be specified in the EEPROM
2. SM0 and SM1 are (generally) used for mailbox communication for a CoE slave
3. Those numbers are not registers, it is object dictionary indexes

Davidinos, 2019-11-07:
Hello mheden,
thank you very much for your answer. I understand the first two points, concerning the third one:
- I just wonder where address 0x6000-0xFFFF are stored. Are they stored in microcontroller memory (considering that LAN9252 has a DRAM memory from 0x1000-0x1FFF)?
- I suppose that SDOs data is exchanged from microcontroller to master through DRAM like it happens with PDOs mapping. What address are used in this case?

I hope that my questions have any sense..
Thank you in advanced.

nakarlsson, 2020-03-09:
1. 0x6000-, are values in the ObjectDictionary part of the host controller application code, not memory addresses.
2. SDOs are exchanged via the mailboxes, the mailboxes are part of the DPRAM.

closed 2020-03-09 by nakarlsson