OpenEtherCATsociety/SOEM #351: Windows Build with MinGW

lemmi25, 2019-10-31:
Hi,

I had some trouble with building on Wndows. I used MinGW.

After running 

```
cmake .. -G "MinGW Makefiles"
```

I got this error:

```
CMake Error at C:/install/Cmake/share/cmake-3.16/Modules/CMakeMinGWFindMake.cmake:12 (message):
  sh.exe was found in your PATH, here:

  C:/install/Git/usr/bin/sh.exe

  For MinGW make to work correctly sh.exe must NOT be in your path.

  Run cmake from a shell that does not have sh.exe in your PATH.

  If you want to use a UNIX shell, then use MSYS Makefiles.

Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
```

after adding:

````
set(DCMAKE_SH="CMAKE_SH-NOTFOUND")
````

to the CMakeLists File the error is gone. Maybe this can be added to the file?

Just in case someone else is getting this error.

Cheers

nakarlsson, 2020-03-08:
Can this handling be added default @hefloryd ?

traversaro, 2020-03-08:
Note that this issue has been fixed in CMake 3.17 that will be released soon (see https://gitlab.kitware.com/cmake/cmake/issues/19327 and https://cmake.org/cmake/help/v3.17/release/3.17.html#other-changes), so the workaround can be made conditional only for CMake < 3.17 .

nakarlsson, 2020-05-09:
@lemmi25 , can you verify the above input?

nakarlsson, 2021-05-28:
inactivity , there are more recent issues on MinGW #513

closed 2021-05-28 by nakarlsson