OpenEtherCATsociety/SOEM #352: Buffer overflow in ecx_receive_processdata_group()

Wanga, 2019-11-07:
At the moment slave disconnected, sometimes weird segmentation fault happen, gdb shows the memory after PDO buffer is corrupted.

To catch who overwrites the memory, we use mmap() to allocate 2 memory pages for PDO map then use mprotect() to let process generate core file when anyone touch second page. PDO size is around 3000 bytes, which is less than 1 page (4096 bytes).

```c
IOmap = (char *)mmap(NULL, ETHERCAT_IOMAP_SIZE * 2, PROT_READ | PROT_WRITE, MAP_ANONYMOUS | MAP_SHARED, -1, 0);
memset(IOmap, 0, ETHERCAT_IOMAP_SIZE * 2);
mprotect(IOmap + ETHERCAT_IOMAP_SIZE, ETHERCAT_IOMAP_SIZE, PROT_NONE);
...
ec_config_overlap_map(IOmap);
```

The backtrace shows PDO receive is the writer for second page.
```
#0  0xf760dba2 in __memcpy_ssse3 () from /lib/libc.so.6
#1  0x00000005 in ?? ()
#2  0xf6f98b14 in ecx_receive_processdata_group ()
   from some_path
#3  0xf6f993b7 in ec_receive_processdata_group ()
   from some_path
#4  0xf6f993fa in ec_receive_processdata ()
   from /some_path
```

So we add this section to check the size we receive. the else part 
```diff
--- a/soem/ethercatmain.c
+++ b/soem/ethercatmain.c
@@ -1934,18 +1934,50 @@ int ecx_receive_processdata_group(ecx_contextt *context, uint8 group, int timeou
          {
             if(first)
             {
-               memcpy(context->idxstack->data[pos], &(context->port->rxbuf[idx][EC_HEADERSIZE]), context->DCl);
-               memcpy(&le_wkc, &(context->port->rxbuf[idx][EC_HEADERSIZE + context->DCl]), EC_WKCSIZE);
-               wkc = etohs(le_wkc);
-               memcpy(&le_DCtime, &(context->port->rxbuf[idx][context->DCtO]), sizeof(le_DCtime));
-               *(context->DCtime) = etohll(le_DCtime);
+               uint32 pdo_len = (context->grouplist[group].Obytes + context->grouplist[group].Ibytes);
+               if((char*)context->idxstack->data[pos]+context->DCl > (char*)pdo_len)
+               {
+                  printf("PDO rx overflow !! PDO range start at addr %p end at %p, size 0x%x\n",
+                          context->slavelist[0].outputs, context->slavelist[0].outputs+pdo_len, pdo_len);
+                  printf("This packet try to write at %p with size 0x%x\n", context->idxstack->data[pos], context->DCl);
+               }
+               else
+               {
+                  memcpy(context->idxstack->data[pos], &(context->port->rxbuf[idx][EC_HEADERSIZE]), context->DCl);
+                  memcpy(&le_wkc, &(context->port->rxbuf[idx][EC_HEADERSIZE + context->DCl]), EC_WKCSIZE);
+                  wkc = etohs(le_wkc);
+                  memcpy(&le_DCtime, &(context->port->rxbuf[idx][context->DCtO]), sizeof(le_DCtime));
+                  *(context->DCtime) = etohll(le_DCtime);
+               }
             else // if(first)
             {
                 /* Add same check */
             }
```

When device disconnect, we got this log, and the segmentation fault doesn't happen.
```
PDO rx overflow !! PDO range start at addr 0xf773f000 end at 0xf773fbda, size 0xbda
This packet try to write at 0xf773fb91 with size 0x5b6
```

It seems a EtherCAT packet is corrupted in this case, however, we didn't have a chance to collect raw packet data by sniffer yet.

Can we add size checking for PDO receive to avoid buffer overflow?

ArthurKetels, 2019-11-07:
Can you print the idxstack when the check fails? 

It is unlikely that a corrupted packed has anything to do with it. The position and size of each memcpy is calulated in ecx_main_send_processdata(). Even with rx packets that are much bigger than expected only the pre-calculated sizes are copied. 

The thing that wonders me is that it is triggered by disconnecting a slave. Two things change, ecx_receive_processdata_group() takes more time (because it will timeout often when no frames are received), and  the recovery thread gets active (when implemented).

Is it possible that the process thread could be overrun by itself? The process data functions are not re-entrant safe. 

If the recovery thread is really aggressive, the process thread could run out of empty buffers and overwrite buffers that are still in use. 

Not withstanding the above, I still can't figure out where in the code the idxstack could be populated with wrong data pointers and/or lengths.

Wanga, 2019-11-08:
I'll try to get log, it may take more than a week.

In our system
- Process thread:  cycling time 1ms , PDO receive timeout 2ms, priority higher
- Recovery thread: cycling time 500ms, priority lower

When the device disconnected, due to priority, I think this overwrite happen before recovery thread has chance to run. Only one process thread is created.
Loop N : wkc = 100
Loop N+1 : wkc = 0, overwrite happen (usually when device lost we got wkc = -1, but this case wkc = 0)
Loop N+2 : wkc = 98, some device lost
... 200ms or more, Recovery thread detect slave lost.

We restart SOEM without terminate process to keep other application running. ec_close() is called, all related thread and data is free, then we use same init flow to start SOEM again.

After restart, everything is fine, but the chance to get this overwrite issue become very high when first slave disconnect.

nakarlsson, 2019-11-11:
What version of SOEM do you use, 1.4.0?

Wanga, 2019-11-11:
Yes, 1.4.0

Log with index, input and input start address.

```
PDO rx overflow !! PDO range start at addr 0xf7787000 end at 0xf7787e8c, size 0xe8c, index 7 pos 1
Input Size 756 Output Size 736, Input Start at 0xf7787736
This packet try to write at 0xf7787cf0 with size 0x5ba
```
0x5ba = 0xf7787cf0 - 0xf7787736

ArthurKetels, 2019-11-11:
I just asked to print the idxstack together with the warning text.

```
/** stack structure to store segmented LRD/LWR/LRW constructs */
typedef struct ec_idxstack
{
   uint8   pushed;
   uint8   pulled;
   uint8   idx[EC_MAXBUF];
   void    *data[EC_MAXBUF];
   uint16  length[EC_MAXBUF];
} ec_idxstackT;

```

Wanga, 2019-11-12:
When PDO needs 2 frames to receive, [if first frame is lost](https://github.com/OpenEtherCATsociety/SOEM/blob/2a8c07c32b53dfcc83f045e2f8e0b51900fa6fb3/soem/ethercatmain.c#L1931), the ["first" flag](https://github.com/OpenEtherCATsociety/SOEM/blob/2a8c07c32b53dfcc83f045e2f8e0b51900fa6fb3/soem/ethercatmain.c#L1921) will keep TRUE to let second frame receiving go to [first frame case](https://github.com/OpenEtherCATsociety/SOEM/blob/2a8c07c32b53dfcc83f045e2f8e0b51900fa6fb3/soem/ethercatmain.c#L1937). which assume using DCl for datalength is always ok

```
ecx_waitinframe got no frame at pos 0 idx 11, wkc 0 wkc2 -1
PDO rx overflow !! PDO range start at addr 0xf7759000 end at 0xf7759e8c, size 0xe8c, index c pos 1
Input Size 756 Output Size 736, Input Start at 0xf7759736
First: This packet try to write at 0xf7759cf0 with size 0x5ba
idxstack pushed 2  pulled 2
pos 0 idx 11 data 0xf7759736 length 1466, rxbufstat 0 
pos 1 idx 12 data 0xf7759cf0 length 412, rxbufstat 0 
pos 2 idx 0 data (nil) length 0, rxbufstat 0 
pos 3 idx 0 data (nil) length 0, rxbufstat 0 
pos 4 idx 0 data (nil) length 0, rxbufstat 0 
pos 5 idx 0 data (nil) length 0, rxbufstat 0 
pos 6 idx 0 data (nil) length 0, rxbufstat 0 
pos 7 idx 0 data (nil) length 0, rxbufstat 0 
pos 8 idx 0 data (nil) length 0, rxbufstat 0 
pos 9 idx 0 data (nil) length 0, rxbufstat 0 
pos 10 idx 0 data (nil) length 0, rxbufstat 0 
pos 11 idx 0 data (nil) length 0, rxbufstat 0 
pos 12 idx 0 data (nil) length 0, rxbufstat 0 
pos 13 idx 0 data (nil) length 0, rxbufstat 0 
pos 14 idx 0 data (nil) length 0, rxbufstat 0 
pos 15 idx 0 data (nil) length 0, rxbufstat 0 

```

Code with debugging log, see last 3 packet in attached file.

[packet.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3834976/packet.zip)

```c
/* get first index */
pos = ecx_pullindex(context);
/* read the same number of frames as send */
while (pos >= 0)
{
   idx = context->idxstack->idx[pos];
   wkc2 = ecx_waitinframe(context->port, context->idxstack->idx[pos], timeout);
   /* check if there is input data in frame */
   if (wkc2 > EC_NOFRAME)
   {
      if((context->port->rxbuf[idx][EC_CMDOFFSET]==EC_CMD_LRD) || (context->port->rxbuf[idx][EC_CMDOFFSET]==EC_CMD_LRW))
      {
         uint32 pdo_len = (context->grouplist[group].Obytes + context->grouplist[group].Ibytes);
         int i;
         if(first)
         {
            if((char*)context->idxstack->data[pos]+context->DCl > (char*)context->slavelist[0].outputs+pdo_len )
            {
               printf("PDO rx overflow !! PDO range start at addr %p end at %p, size 0x%x, index %x pos %x\n",
                       context->slavelist[0].outputs, context->slavelist[0].outputs+pdo_len, pdo_len, idx, pos);
               printf("Input Size %x Output Size %x, Input Start at %p\n", 
                       context->grouplist[group].Ibytes, context->grouplist[group].Obytes, context->slavelist[0].inputs );
               printf("First: This packet try to write at %p with size 0x%x\n", context->idxstack->data[pos], context->DCl);
               printf("idxstack pushed %d  pulled %d\n", context->idxstack->pushed, context->idxstack->pulled) ;
               for(i=0; i<EC_MAXBUF; i++ )
               {
                  printf("pos %d idx %d data %p length %d, rxbufstat %d \n", i, context->idxstack->idx[i], context->idxstack->data[i], 
                                                                             context->idxstack->length[i], context->port->rxbufstat[i]) ;
               }
               exit(-1);
            }
            else
            {
              // ... some copy
            }
            first = FALSE;
         }
         else
         {
            if((char*)context->idxstack->data[pos]+context->idxstack->length[pos] > (char*)context->slavelist[0].outputs+pdo_len )
            {
               printf("PDO rx overflow !! PDO range start at addr %p end at %p, size 0x%x, index %x pos %x\n",
                       context->slavelist[0].outputs, context->slavelist[0].outputs+pdo_len, pdo_len, idx, pos);
               printf("Input Size %x Output Size %x, Input Start at %p\n", 
                       context->grouplist[group].Ibytes, context->grouplist[group].Obytes, context->slavelist[0].inputs );
               printf("Not First: This packet try to write at %p with size 0x%x\n", context->idxstack->data[pos], context->idxstack->length[pos]);
               printf("idxstack pushed %d  pulled %d\n", context->idxstack->pushed, context->idxstack->pulled) ;
               for(i=0; i<EC_MAXBUF; i++ )
               {
                  printf("pos %d idx %d data %p length %d rxbufstat %d\n", i, context->idxstack->idx[i], context->idxstack->data[i], 
                                                                              context->idxstack->length[i], context->port->rxbufstat[i]) ;
               }
            }
            else
            {
				/* copy input data back to process data buffer */
                 memcpy(context->idxstack->data[pos], &(context->port->rxbuf[idx][EC_HEADERSIZE]), context->idxstack->length[pos]);
                 wkc += wkc2;
            }
         }
         valid_wkc = 1;
      }
      else if(context->port->rxbuf[idx][EC_CMDOFFSET]==EC_CMD_LWR)
      {
         // ... some code
      }
      else
      {
         printf("ecx_waitinframe got strange frame at pos %d idx %d\n", pos, idx);
      }
   }
   else
   {
      printf("ecx_waitinframe got no frame at pos %d idx %d, wkc %d wkc2 %d\n", pos, idx, wkc, wkc2);
   }
   /* release buffer */
   ecx_setbufstat(context->port, idx, EC_BUF_EMPTY);
   /* get next index */
   pos = ecx_pullindex(context);
}

ecx_clearindex(context);

```

Is there any reason for this line not using context->idxstack->length[pos] for size?
https://github.com/OpenEtherCATsociety/SOEM/blob/2a8c07c32b53dfcc83f045e2f8e0b51900fa6fb3/soem/ethercatmain.c#L1937

ArthurKetels, 2019-11-12:
Thanks for the debugging! 
Yes this is clearly a bug. The "first" flag should be in the idxstack data. I prepare a patch and let you know.

ArthurKetels, 2019-11-12:
I just patched the main branch and mildly tested the result. Please check if the bug is fixed.

nakarlsson, 2019-11-12:
@ArthurKetels , better to do a PR that can be reviewed.

1. Should 
 it be pushed instead of pull?  https://github.com/OpenEtherCATsociety/SOEM/blob/9ec863594381ca087a7a3e12113b83f1558ae182/soem/ethercatmain.c#L1653
2. Should we remove the context->DCtO and context->DCl from the type ecx_contextt? Used internally.

ArthurKetels, 2019-11-12:
Yes right. Made the patch in the wrong repo. My bad.
1. Yes, just by accident this works, because the DC push is always pos=0.
2. Likely, I did not check if there are other users, but probably not.

Wanga, 2019-11-13:
Thanks, issue is fixed after patch 9ec863594381ca087a7a3e12113b83f1558ae182.

closed 2019-11-13 by Wanga