OpenEtherCATsociety/SOEM #353: IOMap writing and reading sync

frriccar, 2019-11-07:
Hello,

I am using SOEM to control several motors, and I was wondering which is the correct way to retrieve inputs and upload outputs in the IOMap. I'm referring to concurrency problem, I explain better our idea:

- We want to create an EtherCATContext which is in charge of the detection of ethercat errors and also for the process data exchange (basically we have two threads, like in the SOEM red_test example);

- At the same time, we have several "motor" objects that can write new values or retrieve updated process data from the IOMap (for example velocity, position, etc.).

Now, in the red_test example it seems there is no synchronization between the async part (the "for" cycle in which it prints the IOMap inputs and outputs) and the real time thread in which it exchange the process data between Master and Slaves. So, my question is: What happen if the RT Thread send the IOMap on the ethecat bus while a motor is writing one of its output field? And (reverse), what happens if a motor is reading a value from the IOMap input field while a new IOMap is arriving from the bus?
Is there already a standard solution for this concurrency problem?

Thank you very much.

tecodrive, 2019-11-07:
Principle of EtherCAT communication:

IOMap is the IOBuffer in which the PLC output data (PDO) for drives or other slaves are stored.
The "ec_send_overlap_processdata ()" SOEM function reads IOMap and creates raw Ethernet frames based on the Ethercat frame definition. Thereafter, frames or packets are sent over raw Ethernet sockets.

Ethercat frames go in and out bit by bit through each slave. Each slave knows where its own bits for READ and WRTIE are in the frame. Slave read and write bits in real time when going through the Ethercat frame.

When the Ethercat frame completely returns to the NIC, the function "ec_receive_processdata (EC_TIMEOUTRET / * us * /)" inserts the date in IOMap. The PLC can then read the input data (PDO ) from IOMap.

There is no possibility to read or write to the IOMap at the same time because of the principle of ethercat communication.

ArthurKetels, 2019-11-07:
I assume the OP refers to the synchronization between the process data exchange thread and a separate control thread that reads and writes on the IOmap.

The assumption in SOEM is that there is only one RT tread that does both the process data exchange and IOmap manipulation. That way there is no concurrency problem. 

I you do want/need to have multiple threads manipulating the IOmap then your application is responsible for data coherence. There is no perfect solution to this. You could use buffer copies, or FIFO's, or IPC like zeromq, or any arbitration mechanism you like.

frriccar, 2019-11-11:
Hello,

Thank you for your answers.
Yes, I was referring to the issue caught by @ArthurKetels.
Probably I didn't describe very well my doubt.

I will close for now the issue.
Thank you very much.

closed 2019-11-11 by frriccar