OpenEtherCATsociety/SOEM #354: SOEM + slaves with 32-bit DC time representation

Dmitriy-V-Ivanov, 2019-11-09:
The function ecx_configdc() calculates current DC time (mastertime64) and sets it to all drives.

Issue 1.

When using slaves with 32-bit DC time representation, we must handle time overflows:

// Init
int64 ref_dc_time = 0;
int64 last_dc_time_sample = 0;

// Call in cycle
void KeepDCTime()
{
  uint32 hi = (uint32)(ref_dc_time >> 32);
  uint32 lo = (uint32)ec_DCtime;
  uint32 last_lo = (uint32)last_dc_time_sample;
  if (lo < last_lo) { hi++; }
  ref_dc_time = ((int64)hi << 32) | lo;
  last_dc_time_sample = ec_DCtime;
}

This generally works, but what happens if low 32-bit of mastertime64 is close to its max value?

If an overflow of 32-bit DC time occures between call of ecx_configdc() and first call of KeepDCTime(), THIS OVERFLOW WILL BE LOST.

This will result in an incorrect DC time value.

Its desired to set mastertime64 = 0 for 32-bit DC time representation. Then there will be a resource of time of 4 seconds before first call of KeepDCTime().

Issue 2.

It seems that for slaves with 32-bit DC time representation (Delta ASDA A2-E), the function ecx_configdc() sets DC time as (mastertime64 & 0xFFFFFFFF), rather that mastertime64.

In the case when slaves with 32 and 64-bit runs together, we have two different DC times in group (slaves will have different sync points)?

Setting mastertime64 = 0 in ecx_configdc() will also help here.

Dmitriy-V-Ivanov, 2019-11-10:
For slaves with 32-bit DC time representation, applying 64-bit current master time will corrupt it (in terms of sync points):

(mastertime64 % CyclTime) != ((mastertime64 & 0xFFFFFFFF) % CyclTime)

To fit current time to 32-bit word, we neede to reduce it by number of full cycles before applying to slave:

mastertime64 % CyclTime

Then current time remains unchanged in terms of sync points.

Dmitriy-V-Ivanov, 2019-11-10:
What is proposed to do.

Allow user to pass custom value of current DC time to ecx_configdc().

Something like:

boolean ecx_configdc2(ecx_contextt *context, uint64 mastertime64)

How is it supposed to be used:

// assign full 64-bit DC time (or 0, if desired)
int64 ref_dc_time = <current DC time>;

// assign reduced DC time to fit 32-bit limitation
int64 last_dc_time_sample = ref_dc_time % CyclTime;

// pass reduced DC time as initial DC time;
ecx_configdc2(&ecx_context, last_dc_time_sample);

...

// Keep DC time In cycle
ref_dc_time += (uint32)ec_DCtime - (uint32)last_dc_time_sample;
last_dc_time_sample = ec_DCtime;

Dmitriy-V-Ivanov, 2019-11-10:
Or simplified:

int64 ref_dc_time = 0;
int64 last_dc_time_sample = 0;

ecx_configdc2(&context, 0);

...

// Keep DC time In cycle
ref_dc_time += (uint32)(*context.DCtime) - (uint32)last_dc_time_sample;
last_dc_time_sample = *context.DCtime;

ArthurKetels, 2019-11-11:
I would suggest to stay away from slaves with 32bit DC. And if no other option, put a 64bit DC slave as the first in the line. Messing with 32bit clocks is a p.i.t.a.

Dmitriy-V-Ivanov, 2019-11-12:
Thank you, I agree regarding mix 32-bit and 64-bit DC time devices.

But a simple stand-alone system with all 32-bit DC time drives works fine, except for unreliable initialization due to the possible loss of overflow between ecx_configdc() and KeepDCTime() calls.

The problem, of course, can be easily solved by editing ecx_configdc():

mastertime64 = 0;

But it's a little unaesthetic.

closed 2019-11-12 by Dmitriy-V-Ivanov