OpenEtherCATsociety/SOEM #355: Slave can not switch to Pre-op

casinia, 2019-11-11:
During the process of initializing the slave, sometimes the slave cannot go from the Init state to the Pre-op state, but sometimes can.

This is the packet I grabbed under abnormal conditions.
[not into pre-op.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3830361/not.into.pre-op.zip)
I also printed some information.
Slave:1
 State: 1
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 5a65726f ID: 00029252 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00010064 Type:3
 SM3 A:1400 L:   0 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 2f FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

Slave:2
 State: 1
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 5a65726f ID: 00029252 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00010064 Type:3
 SM3 A:1400 L:   0 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 2f FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

Slave:3
 State: 1
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1003
 Man: 5a65726f ID: 00029252 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00010064 Type:3
 SM3 A:1400 L:   0 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 2f FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

Slave:4
 State: 1
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1004
 Man: 5a65726f ID: 00029252 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00010064 Type:3
 SM3 A:1400 L:   0 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 2f FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

Slave:5
 State: 1
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1005
 Man: 5a65726f ID: 00029252 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00010064 Type:3
 SM3 A:1400 L:   0 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 2f FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

Slave:6
 State: 1
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1006
 Man: 5a65726f ID: 00029252 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00010064 Type:3
 SM3 A:1400 L:   0 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 2f FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

ArthurKetels, 2019-11-11:
The problem is not SOEM but your 3rd slave. It has a bug that it forgets to read the ALcommand register after change to INIT when you do this more than once. See line 702 in your wireshark capture. In the slave firmware EACH write to 0x120 by the master should be acknowledged by the slave trough reading of 0x120.

casinia, 2019-11-12:
But will this cause all the slaves to switch to BOOTSTRAP?

ArthurKetels, 2019-11-12:
No this is an observation effect. SOEM uses a BRD to read the ALstatus register (0x130) from all slaves at once. Each slave puts it register data with logic OR to the result. Most slaves answer with 0x02 (pre-OP), but one slaves answers with 0x01 (Init). The result is 0x01 OR 0x02 = 0x03. It is a coincidence this corresponds to state Bootstrap 0x03. 

Compare wireshark line 140 and 142. Both are the result of a BWR to 0x120 (ALcommand). So both should return the same WKC. But line 140 returns the result of 6 slaves where line 142 returns only 5. So evidently one slave does not answer. This turns out to be slave 3. A write from the master to 0x120 is blocked as long the slave firmware does not acknowledge the state change (by reading 0x120).

From the slave conformance documentation it is mandatory for all slaves to be able to be put to Init mode at ALL times. Clearly the slave is violating this rule. So please correct the slave firmware, and you will see this problem is gone.

nakarlsson, 2020-03-09:
Seems inactive so we'll close

closed 2020-03-09 by nakarlsson