OpenEtherCATsociety/SOEM #357: A slave can not reach operational state with No error

zhuowen, 2019-11-28:
1 slaves found and configured.
Found ECAT-2011H 16S.E./8Diff Ch. Ana. at position 1
ec_config_map_group IOmap:00902200 group:0
 >Slave 1, configadr 1001, state 02
  CoE Osize:0 Isize:512
     ISIZE:512 512 OSIZE:0
  SM programming
    SM2 Type:3 StartAddr:1100 Flags:00000064
    SM3 Type:4 StartAddr:1400 Flags:00010020
 =Slave 1, INPUT MAPPING
    FMMU 0
      SM3
    Inputs 00902200 startbit 0
IOmapSize 64
Slaves mapped, state to SAFE_OP.

ERROR No error !!!!!!!!!!!!

segments : 1 : 64 0 0 0
Request operational state for all slaves
Calculated workcounter 1
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error

[bag.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3901652/bag.zip)

zhuowen, 2019-11-28:
[ICPDAS ECAT-2011.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3901660/ICPDAS.ECAT-2011.zip)

ArthurKetels, 2019-11-28:
Nothing looks out of the ordinary. Could you tryy a timeout value safe-OP->OP of more than 2s.
From the ESI file
```
-<Timeout>
    <PreopTimeout>2000</PreopTimeout>
    <SafeopOpTimeout>9000</SafeopOpTimeout>
    <BackToInitTimeout>5000</BackToInitTimeout>
    <BackToSafeopTimeout>200</BackToSafeopTimeout>
</Timeout>
```
So a 9s timeout would be more appropriate.

zhuowen, 2019-12-02:
> Nothing looks out of the ordinary. Could you tryy a timeout value safe-OP->OP of more than 2s.
> From the ESI file
>
> ```
> -<Timeout>
>     <PreopTimeout>2000</PreopTimeout> [...]

Thank you,  I have tried set the timeout 9s, but it still not work. Attatch is Wireshark with TwinCAT, thanks again.
[bags.zip](https://github.com/OpenEtherCATsociety/SOEM/files/3909982/bags.zip)

ArthurKetels, 2019-12-02:
Just for test can you add
ecx_context.slavelist[1].FMMUunused = 1;
just before
ec_config_map(&IOmap);
?

The only important difference I can see between SOEM and Twincat is that Twincat reserves FMMU0 for outputs and programs the inputs on FMMU1. SOEM sees there are no outputs and uses FMMU0 for inputs. Increasing FMMUunused from 0 to 1 will force the use of FMMU1 for inputs.

nakarlsson, 2020-03-03:
@zhuowen , still an issue?

closed 2020-03-09 by nakarlsson