OpenEtherCATsociety/SOEM #358: Ethercat GateWay IO Module PDO Config

Bingyu1413, 2019-11-29:
Recently, I used a gateway IO Module, The gateway is Ethercat, and the child io module used RS485 to communicate with gateway.

The gateway's pdo can't right config.I've tried TWINCAT/CODESYS to config the gateway,It works fine.
When I use soem to config the pdo in PO2SOconfig,I found the 0x1c12 and 0x1c13 can't be write, from the  gateway's xml,0x1c12 and 0x1c13 is truly readonly.

I confused how to config the gateway like TWINCAT or CODESYS.
Could you give some advise to help me with this proplem?

Bingyu1413, 2019-11-29:
here is the xml file of the gateway.

[ELCO_ECAT_SPEC_GW_001 - 副本.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3903269/ELCO_ECAT_SPEC_GW_001.-.txt)
[ELCO_ECAT_SPEC_GW_Modules - 副本.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3903270/ELCO_ECAT_SPEC_GW_Modules.-.txt)

ArthurKetels, 2019-11-30:
Can you post two wireshark captures of the cinfiguration phase? One with twincat and one with soem?

When 0x1c12 and 0x1c13 are RO it means the slave itself will populate the proper subindexes according to the modules connected. 

One common problem found is that not all slaves with CA(complete access) support reading 0x1c12 and 0x1c13 with the CA bit set. Setting this bit to zero in the slave tree will then fix the problem.

Bingyu1413, 2019-12-02:
I have configured the gateway as the bellow code when slave->PO2SOconfig = Test_Setup:        

u8val = 0x00;
	ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTSAFE);

	u32val = 0x1A00;
	ec_SDOwrite(slave, 0x1c13, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTSAFE);

	u32val = 0x1A01;
	ec_SDOwrite(slave, 0x1c13, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTSAFE);

	u32val = 0x1A02;
	ec_SDOwrite(slave, 0x1c13, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTSAFE);

	u32val = 0x1A03;
	ec_SDOwrite(slave, 0x1c13, 0x04, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTSAFE);

	u8val = 0x04;
	ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTSAFE);

ArthurKetels, 2019-12-02:
First, it is of no use to try to write to read-only objects. Second, you have to use the correct datatype. 0x1c13 entries are 16 bits not 32 bits. Third, you could do with checking the error status after each ec_SDOwrite. SOEM has a nice function ecx_elist2string to print a readable text after an error.

Could you show me the result of :
slaveinfo eth? -map
and
slaveinfo eth? -sdo
(replace eth? with the name of your enternet port)

Bingyu1413, 2019-12-03:
Here is the -map and -sdo info.
[-map.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3913887/-map.txt)
[-sdo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3913888/-sdo.txt)

And the child io module is:
[0]SPDB-0800D-013
[1]SPDB-08UP-011
[2]SPDB-0008D-011
[3]SPDB-08UP-011

ArthurKetels, 2019-12-03:
I do not have the wireshark capture I asked for, so answers on details are a bit difficult to give. But it looks like SOEM does everything OK. 

From your slaveinfo you can see that the slave even goes to safe-OP without problem. This means the slave thinks the configurations is in order.

What is noteworthy that it looks like the first module (0800D-013) is not in the mapping list. I can only see slots 1, 2 and 3. Modules seem to start with 0x6001 and 0x7001 instead of 0x6000 and 0x7000. This could perhaps signal a firmware bug in the gateway slave.

Unfortunately the -sdo print shows only one line. So it seems the dictionary is not functional in the gateway slave.

And without the wireshark files I can not do a more thorough investigation.

Bingyu1413, 2019-12-04:
Thanks for your help,I will get the wireshark capture later.

ArthurKetels, 2019-12-04:
Thanks for the files. 

As expected nothing wrong with the codesys capture. Typical configuration from ENI file. Init, check if slave is the same as in ENI file. Straight programming of SM and FMMU. Then startup of process data exchange and going from safe-OP to OP.

SOEM does not read the ESI file but instead has to find out the configuration and mapping from the slave itself. And although the slave is not very efficient in supplying the data (many empty slots with zero data), the configuration of SM and FMMU succeeds. The slave goes to safe-OP.

Then unfortunately you selected the wrong program to test if the slave goes to OP. Slaveinfo only goes as far as safe-OP and then request the object dictionary as requested by the -sdo flag. The slave then makes a mess of things and returns invalid CoE responses. 

Can you do a proper test and try the simple-test program? And please make a wireshark capture of that event.

Bingyu1413, 2019-12-05:
I'll try it later,and give a Wireshark capture file,Thanks for your patience reply extremely.

Bingyu1413, 2019-12-10:
Here is the wireshark capture file of simple_test.

ArthurKetels, 2019-12-10:
As expected SOEM does work as intended. The slave goes to OP. 

The difference between the Codesys and SOEM run is the PDO length. In Codesys (correct) it should be 3 outputs and 7 input bytes. SOEM configures 3 outputs and 5 inputs.

The problems are in the slave firmware as it skips the first module information on the SDO interface.

> Modules seem to start with 0x6001 and 0x7001 instead of 0x6000 and 0x7000.
  
The two input bytes from 1st module are skipped.

The solution is to ask the slave vendor to update its firmware. And while at it ask them to solve the other bugs too.

Bingyu1413, 2019-12-10:
Now I know SOEM is diffrent from CODESYS or TWINCAT(...etc) in some place.

I'd like to know is there any way to make soem work like CODESYS?
Such as config pdo with ESI file.

ArthurKetels, 2019-12-10:
Yes you can use SOEM as Codesys. 

Simply do not use ec_config_map(), and replace it with your ENI parser. Then write the SM anf FMMU directly with the data found in the ENI file.

Others have done this with success.

Still, the slave firmware has bugs, and should be solved.

Bingyu1413, 2019-12-10:
Thank you for your help and advice.
 I'll try to write SM and FMMU to make this slave work fine.

nakarlsson, 2020-03-08:
No such sample, you can check the ethercatconfig.c for standard SOEM config if SM/FMMU.

closed 2020-03-08 by nakarlsson