OpenEtherCATsociety/SOEM #361: SOEM stops randomly

yoyixms2, 2020-01-10:
Hello,
I have written a program based on the simple test to run a master. For some unknown reason, the program at some random point stops doing a transfering proper data, without actually crashing. this can happen after a few seconds or after more than an hour. I tried then only with a simple test demo program, modified only to run for a longer time, and here the problem occurs as well. I tried to debug it, but i could not find the source of this problem. What also happens after this problem occurs is that i cannot restart the simple_test program because it can´t connect anymore. A computer restart solves this problem. 
I am using a windows 10 PC, with npcap in winpcap compatible mode. the connected slaves are the EK1100 coupler + some other modules.
thanks

ArthurKetels, 2020-01-11:
Win10 problem. Do not assume windows can be used for any serious control application. Timing is all over the place and sometimes resources (like network) can disappear without warning.

Things that can help is disabling power saving. If you have a laptop, use performance mode. Look into your NIC driver settings if some power saving or sleep mode can be disabled.

yoyixms2, 2020-01-14:
I disabled all power saving options from CPU and NIC and enabled performance mode, but it is still happenning. I tested with TwinCAT in config mode and seems to run fine. Any reason for that?

ArthurKetels, 2020-01-14:
TwinCAT uses a real time extension. This is like a second OS next to windows. Only this way they can guarantee performance. What you see on your screen is only a GUI that communicates via IPC to the real time kernel.

SOEM simply compiles on windows itself. And therefore is depending on the services delivered by windows. If this is not satisfactory, switch to another OS. Linux with preemptRT is working satisfactory for the majority of use cases. For stable PDO performance over 20Kcycles/s with SOEM I would advice to use a bear metal approach. Even a simple ARM cortex M4 can handle this with ease.

yoyixms2, 2020-01-21:
I tried with a different computer (windows 10 desktop PC) and it has been running for about a day without any problems.

closed 2020-01-21 by yoyixms2