OpenEtherCATsociety/SOEM #362: SOEM Communication Error When Switching to Safe-OP from Pre-Op

elk1984, 2020-01-16:
Hello,

I have used SOEM successfully with servo drives from Panasonic and Yaskawa, but for some reason when using the Mitsubishi EtherCAT drive I am getting a communication error when giving the command to transition from PreOp to SafeOp. Is there something in the ESI file that would tell me why I am unable to transition from PreOp to SafeOp? The Mitsubishi drive uses the Anybus M40 module from HMS technologies, I'm not sure if SOEM has been used with this in the past. Any help would be appreciated! I'm unable to upload the ESI xml file here because of the file type...

Thanks,

ArthurKetels, 2020-01-17:
Post a wireshark capture. Mostly it will tell the reason for failure.
P.s. when you zip the xmf file you should be able to post it here.

elk1984, 2020-01-17:
I have attached the xml and the wireshark, I see the drive go into an error
state when the command is given to switch into safeop from preop...I don't
see where in the wireshark this is happening though

elk1984, 2020-01-17:
[mitsu_wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4077734/mitsu_wireshark.zip)

I have attached the xml and the wireshark, I see the drive go into an error state when the command is given to switch into safeop from preop...I don't see where in the wireshark this is happening though

elk1984, 2020-01-17:
To be more specific, I see at packet 38398 the command to switch into SAFEOP, shortly after the BRD's show PREOP change into PREOP+ERROR at packet 38415. The error returned is x12, which I'm not sure if this maps to the AL Status codes, but if it does I believe means "unknown requested state change". Hopefully that helps...

ArthurKetels, 2020-01-17:
Thanks for the files. It makes it clear why the slave will not go to safeop. SM2 and SM3 activate bit is set to zero. This means no SM for outputs and inputs. Of course the checks the slave make before going into safeop will fail. 

The SM activate bits are set wrong in both the eeprom and the ESI file. This is a clear bug in the slave data files.

But as of commit 4427684cc553d3b8a2b6c675693097cc12a424e4 SOEM will set the activate bit anyhow when it detects a non zero size SM. So please update SOEM to the latest version.

Second thing I noticed is that the complete access feature of the slave is broken. It works but very slow and inefficient (sending tons of zero bytes).  Remove the CA flag in the slave settings in your hook function.

elk1984, 2020-01-17:
Hi Arthur,

Just by adding that section of code regarding the SM activate bits worked
like a charm!! Thank you very much.
As far as the complete access feature being broken for the slave, I went
through the code and I always had CA set to FALSE for SDO reads and
writes...I'm not sure what was causing the behaviour you are mentioning.
Further in the code the only places I see CA being set to true are in
ethercatconfig.c with a call to ecx_readPDOmapCA()  and in
ecx_readPDOassignCA() in ethercatcoe.c...but I don't think these are
getting called...

Anywho, the drive is in the OP state so I'm happy.

Thanks,

nakarlsson, 2020-03-08:
Seems fixed

closed 2020-03-08 by nakarlsson

elk1984, 2020-03-08:
Yes thanks