OpenEtherCATsociety/SOEM #366: delay error in red_test

pvn1420297, 2020-02-04:
hello,
i am using ubuntu 18.04. i tried to test slaveinfo and simple_test and red_test with RTA drive. Slaveinfo & Simple_test both are working fine. In Red_test, i did little bit modification to run motor. but there is  delay (0.1).

output-

SOEM (Simple Open EtherCAT Master)
Redundancy test
Starting Redundant test
ec_init on enp7s0 succeeded.
1 slaves found and configured.
Slave:1 Name:RTA X-PLUS ET  EtherCAT Output size: 64bits Input size: 64bits State: 4 **delay:0.1**
         Out:cac9f1a0,   8 In:cac9f1a8,   8
Calculated workcounter 3

ArthurKetels, 2020-02-04:
Hi, if you would have taken the effort to look at the code you would know it prints the propagation delay in ns with respect to the reference slave and after the dot if the slave is DC capable. In this case since it is the first slave it has zero delay (with itself), and has DC capability.

So all is fine with your slave.

pvn1420297, 2020-02-05:
thanks for the clarification.

closed 2020-02-05 by ArthurKetels