OpenEtherCATsociety/SOEM #367: Redundancy case not working and causing lost of all slaves

requinham, 2020-02-05:
Hello,
I create a program on Linux using 4 slaves (2 netX and 2 EasyCAT).
I activate redundancy by cabling my last slave to the second ethernet interface of my SOEM master (which is runned on Debian 10 on PC).
When I remove cable from second ethernet interface and last slave (line topology), system works for a few seconds and stop with error saying that all slaves is lost !
I don't have this problem when I disconnect cable between main ethernet interface and first slave for example or between two slaves.
Is this a known issue on SOEM ?
Thanks

nakarlsson, 2020-02-05:
I don't think it is an SOEM issue since it works for a while, or is is just a delayed error reporting? Can you take a wireshark when running the failure scenario?

requinham, 2020-02-05:
Please found below two scan.
Which timeout I need to increase?

[Compressed Trace.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4159769/Compressed.Trace.zip)

nakarlsson, 2020-02-05:
What application are you running? linux\red_test.c? 
Can you add a console output and a capture of the other interface?

requinham, 2020-02-05:
No with another program but with red_test, I have this output :
When connecting two master interfaces with slaves on ring (redundancy) M1==S1==S2==S3==S4==M2, I start red_test and the program say : No slaves found! ... End program
When connecting only M1 without M2 (no ring cable connection), program works and detect all slaves but when (without stopping) I connect the second Master port to last slave, I get this output:
ERROR : slave 1 lost052 , Wck -1, DCtime .  3390028496
ERROR : slave 2 lost054 , Wck -1, DCtime .  3390028496
ERROR : slave 3 lost056 , Wck -1, DCtime .  3390028496
ERROR : slave 4 lost057 , Wck -1, DCtime .  3390028496
and continue displaying Wck at -1 until I disconnect cable between second ethernet port in master and last slave.
any idea.

nakarlsson, 2020-02-06:
Ok, first try to solve why "No slaves found". Take 2 x wireshark , one on each interface when alla cables connected and the master should find slaves.

nakarlsson, 2020-03-08:
Inaktive, can we close this?

requinham, 2020-03-08:
I don't found any solution in my case but Yes you can close this ticket since this occur just for me.

closed 2020-03-08 by nakarlsson

wmyqcy, 2024-11-06:
您好，我现在使用SOEM 1.4 遇见的问题跟你一样，我在linux系统上运行 linux\red_test.c，程序提示说没有找到从站，程序就停止了。我自己参考red_test.c编写了一个冗余程序，可以扫描到2个从站，电机也可以旋转，当我断开EtherCAT主站和从站的网线时，电机就停止旋转了，当我断开2个从站的网线，第1个从站就停止旋转了，第二个从站正常旋转，是不是冗余功能有BUG？

wmyqcy, 2024-11-06:
是不是冗余功能有BUG？